wasm-micro-runtime/wasm-micro-runtime #3063: error when 'try_run' is done in a cross compilation cmake context

ThomasDevoogdt, 2024-01-21:
@wenyongh, this creates an error for cross compilation targets, can you make this try_run conditionally?
See https://cmake.org/cmake/help/latest/command/try_run.html#behavior-when-cross-compiling.

_Originally posted by @ThomasDevoogdt in https://github.com/bytecodealliance/wasm-micro-runtime/issues/2437#issuecomment-1902714929_

wenyongh, 2024-01-22:
@ThomasDevoogdt could you try whether the following patch works for you?
```diff
diff --git a/build-scripts/config_common.cmake b/build-scripts/config_common.cmake
index 76ebcb0b..31894e51 100644
--- a/build-scripts/config_common.cmake
+++ b/build-scripts/config_common.cmake
@@ -425,7 +425,8 @@ if (WAMR_DISABLE_WRITE_GS_BASE EQUAL 1)
   add_definitions (-DWASM_DISABLE_WRITE_GS_BASE=1)
   message ("     Write linear memory base addr to x86 GS register disabled")
 elseif (WAMR_BUILD_TARGET STREQUAL "X86_64"
-        AND WAMR_BUILD_PLATFORM STREQUAL "linux")
+        AND WAMR_BUILD_PLATFORM STREQUAL "linux"
+        AND (NOT CMAKE_CROSSCOMPILING))
   set (TEST_WRGSBASE_SOURCE "${CMAKE_BINARY_DIR}/test_wrgsbase.c")
   file (WRITE "${TEST_WRGSBASE_SOURCE}" "
   #include <stdio.h>
```

ThomasDevoogdt, 2024-01-22:
> @ThomasDevoogdt could you try whether the following patch works for you?

Hi, I've tested this, and it indeed works. Just a question tough, this now means that WASM_DISABLE_WRITE_GS_BASE is not set, so in other words that WRITE_GS_BASE is enabled. I don't know what WRITE_GS_BASE does, and if enabling it is the right default. So it would be nice if you could clarify that.

I tested the compilation on a few toolchains, making use of the buildroot infrastructure (used the fluent-bit package):
```
                    bootlin-x86-64-glibc [24/45]: OK
                     bootlin-x86-64-musl [25/45]: OK
                   bootlin-x86-64-uclibc [26/45]: OK
                   br-i386-pentium4-full [31/45]: OK
                br-i386-pentium-mmx-musl [32/45]: OK
```

wenyongh, 2024-01-22:
@ThomasDevoogdt thanks for the feedback, the feature is trying to improve the performance of AOT and LLVM JIT, please refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#disable-the-writing-linear-memory-base-address-to-x86-gs-segment-register
and
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/perf_tune.md#3-enable-segue-optimization-for-wamrc-when-generating-the-aot-file

BTW, I uploaded PR #3066 to enhance the build script, could you please help check again?

ThomasDevoogdt, 2024-01-22:
> BTW, I uploaded PR #3066 to enhance the build script, could you please help check again?

The compilation happily succeeds.

closed 2024-01-24 by ThomasDevoogdt