wasm-micro-runtime/wasm-micro-runtime #3087: Exception support: Compatibility with LLVM 17

fredldotme, 2024-01-26:
# The problem:

WAMR with its current exception-handling branch has some support for it but various things happened in the meantime: new proposals getting made, and probably implemented.

So much so that the current LLVM 17 upstream stable is unable to function properly as a compilation toolchain for exception-handling support, not even with this fork of it: https://github.com/fredldotme/llvm-project

# How to test:

- Build and check out this repository on Linux or macOS: https://github.com/fredldotme/Tide
- Manual intervention: Enable `EXCEPTION=ON` in `bootstrap.sh`
- Bootstrap the environment with this script

Invoking `bash bootstrap.sh` on macOS should then lead you to getting a full Clang environment set up that is able to build WASI using CMake (without Emscripten).

# The status quo:

Right now running two binaries creates two different results, both failing in proper exception handling:

- `{ printf("Hello\n"); throw; }` results in an exception from the WAMR environment rather than the WASM binary
- More sophisticated C++ linking against libcxx{,abi} might cause it to get code emitted which has `unknown label` errors thrown at WAMR runtime.

wenyongh, 2024-04-26:
Hi, @yamt, I found that you wanted to help look into this issue (https://github.com/bytecodealliance/wasm-micro-runtime/issues/1884#issuecomment-1916055375) and had submitted several PRs before, eg. #3106, #3107, not sure whether the issue is resolved now? Thanks.

yamt, 2024-04-26:
> Hi, @yamt, I found that you wanted to help look into this issue ([#1884 (comment)](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1884#issuecomment-1916055375)) and had submitted several PRs before, eg. #3106, #3107, not sure whether the issue is resolved now? Thanks.

it certainly has a better chance to work than before.
at least i'm sure trivial cases compiled with LLVM 17 work.

@fredldotme can you confirm if/when you have time? thank you.

fredldotme, 2024-04-28:
I have indeed been able to compile and run code with exceptions on my Clang iPad enablement branch and execute it using WAMR on the tablet itself! Great job! I was wondering about more sophisticated test cases but when I encounter them I surely would like to report them.