wasm-micro-runtime/wasm-micro-runtime #3091: [RFC] Basic support for Memory64 proposal

TianlongLiang, 2024-01-26:
# Summary

Plan to implement basic support for the memory64 proposal, first focusing on the classic interpreter and AOT running modes.

## Overview of Basic Support Plan

With basic support completed, users should experience the following:

- Use `cmake -DWAMR_BUILD_MEMORY64=1` to compile a memory64-enabled iwasm VM core. Initially, only memory64-enabled wasm/aot files will be supported. In other words, a memory64-enabled runtime VM core will only support memory64 wasm/aot files for basic support; running memory32 wasm/aot files is not possible (though you can still run memory32 wasm/aot files with the default runtime compile without adding the cmake flags). Further development will ensure compatibility of memory64-enabled runtime with both memory32 and memory64 wasm/aot files.

- For the wamrc AOT compiler, no extra work is required during compile time. To compile a memory64 aot file, use `wamrc --enable-memory64 -o test.aot test.wasm` to generate a memory64 aot file.

  > PS: A warning/exception may be given if the memory64 feature is not actually used in a memory32 wasm file.

- When embedding a memory64-enabled WAMR VM core into your host application and using memory64 wasm/aot files, new APIs (to be discussed) are required to be used for managing linear memory and converting/checking between host native and app linear memory addresses. These APIs are like `wasm64_xxxx` instead of the old `wasm_xxxx` versions:
  
    ```C
    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN uint64
    wasm64_runtime_module_malloc(WASMModuleInstanceCommon *module_inst, uint64 size, void **p_native_addr);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN void
    wasm64_runtime_module_free(WASMModuleInstanceCommon *module_inst, uint64 ptr);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN uint64
    wasm64_runtime_module_dup_data(WASMModuleInstanceCommon *module_inst, const char *src, uint64 size);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN bool
    wasm64_runtime_validate_app_addr(WASMModuleInstanceCommon *module_inst, uint64 app_offset, uint64 size);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN bool
    wasm64_runtime_validate_app_str_addr(WASMModuleInstanceCommon *module_inst, uint64 app_str_offset);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN bool
    wasm64_runtime_validate_native_addr(WASMModuleInstanceCommon *module_inst, void *native_ptr, uint64 size);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN void *
    wasm64_runtime_addr_app_to_native(WASMModuleInstanceCommon *module_inst, uint64 app_offset);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN uint64
    wasm64_runtime_addr_native_to_app(WASMModuleInstanceCommon *module_inst, void *native_ptr);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN bool
    wasm64_runtime_get_app_addr_range(WASMModuleInstanceCommon *module_inst, uint64 app_offset, uint64 *p_app_start_offset, uint64 *p_app_end_offset);

    /* See wasm_export.h for description */
    WASM_RUNTIME_API_EXTERN bool
    wasm64_runtime_get_native_addr_range(WASMModuleInstanceCommon *module_inst, uint8 *native_ptr, uint8 **p_native_start_addr, uint8 **p_native_end_addr);
    ```

## Impact of memory64 proposal on WAMR's data structure

This section list the changes in memory64 proposal that could potentially influence WARM's internal data structure and code logic. Let's see what those influence coulde be

- Memory initial/max page count: According to memory64 proposal, when memory64 are used, the default page size remains unchanged(64KB), but the initial page count/max page count theoretically can maximally be 2^48 now, which is roughly 2.2 millions terabytes. Given it's kind of surreal to use that much of memory in real life, stick with the uint32 representation of page count.
  
  To conclude, the data structure represent WASMMemory **stay unchanged**, the initial page count and max page count are stay as u32.

  > PS: The maximum value can be 2^32 - 1 pages which is about 275 TB. The value exceeding the value is marked by a throw exception. It's not fully conform to spec but it's practical design choice.

- Data segment offset: for memory64, it should be should be i64 now
  - In `WASMDataSeg`, the offset's type of will become i64

- **Memory instruction**: The type of memarg’s offset in all memory instructions needs to be changed, as well as the parameters representing linear memory addresses taken by the memory instruction on the stack, a total of 116 opcodes.

  - In `WASMModule` and `AOTModule` the fields that represent linear memory address:

    ```C
    /* the index of auxiliary __data_end global,
      -1 means unexported */
    uint32 aux_data_end_global_index;
    /* auxiliary __data_end exported by wasm app */
    /* TODO: type changes to uintptr_t */
    uint32 aux_data_end;

    /* the index of auxiliary __heap_base global,
      -1 means unexported */
    uint32 aux_heap_base_global_index;
    /* auxiliary __heap_base exported by wasm app */
    /* TODO: type changes to uintptr_t */
    uint32 aux_heap_base;

    /* the index of auxiliary stack top global,
      -1 means unexported */
    uint32 aux_stack_top_global_index;
    /* auxiliary stack bottom resolved */
    /* TODO: type changes to uintptr_t */
    uint32 aux_stack_bottom;
    ```

    Also in AOT file, those fields will be emitted as u64. So in a sense, AOT ABI will change.

  - In `WASMMemoryInstance(AOTMemoryInstance)`

    ```C
    /* TODO: changes to is_memory_64 flags */
    uint8 __padding__;

    /* TODO: type changes to uint64 */
    uint32 memory_data_size;

    ......other data member

    /* TODO: maybe unify sw boundary check? */
    MemBound mem_bound_check_1byte;
    MemBound mem_bound_check_2bytes;
    MemBound mem_bound_check_4bytes;
    MemBound mem_bound_check_8bytes;
    MemBound mem_bound_check_16bytes;
    ```

    memory_data_size type changes can result to AOT offset calculation logic changes(details are in `create_memory_info`)

    The MemBound is used for software boundary check, its type is uniptr_t. The initial plan is to make it uint64 and unify the software boundary check code logic so that make the corresponding code cleaner. Not sure how much will impact performance for 32-bit platform, if not too much, the initial plan will be implemented.

## Topics for Further Discussion

😄 Suggestions or input on the following topics (or others you can think of) are appreciated.

### 1. Should Memory64 Only Be Supported on 64-Bit Machines?
  
I believe it's not very useful to support the memory64 proposal on 32-bit machines, as more than 4GB of memory can't be utilized anyway. If linear memory is below 4GB, sticking with default memory32 seems sensible. Supporting memory64 on 32-bit machines would also add implementation complexity. **Therefore, the initial plan is not to support the memory64 proposal on 32-bit machines.**

### 2. Exported Runtime APIs for Host Use When Embedding WAMR VMCore
  
As previously mentioned, the **initial plan is to add new APIs like `wasm64_xxx`** for memory64 wasm/aot files, while keeping the old APIs unchanged for memory32 wasm/aot files.

```C
// New APIs
WASM_RUNTIME_API_EXTERN uint64
wasm64_runtime_module_malloc(WASMModuleInstanceCommon *module_inst, uint64 size, void **p_native_addr);
// Old counterpart APIs
WASM_RUNTIME_API_EXTERN uint32
wasm_runtime_module_malloc(WASMModuleInstanceCommon *module_inst, uint32 size, void **p_native_addr);
```

Alternatively, we could modify existing APIs without adding new ones.

```C
// Choose 1: use uintptr_t
WASM_RUNTIME_API_EXTERN uintptr_t
wasm_runtime_module_malloc(WASMModuleInstanceCommon *module_inst, uintptr_t size, void **p_native_addr);
// Choose 2: use uint64
WASM_RUNTIME_API_EXTERN uint64
wasm_runtime_module_malloc(WASMModuleInstanceCommon *module_inst, uint32 size, void **p_native_addr);
```

These two choices are worth considering. The first fits well on 32-bit platforms, and the second maintains consistent function signatures. At first glance, they seem preferable to adding new `wasm64_xxx` APIs. However, they could cause problems for existing programs that want to be compiled with the updated version of WAMR, which is why adding new APIs is the primary choice. Additionally, new APIs can reduce programming errors and enhance readability by clearly indicating the memory standard used for a given WASM file/module, with explicit return value types.

loganek, 2024-01-30:
> running memory32 wasm/aot files is not possible (though you can still run memory32 wasm/aot files with the default runtime compile without adding the cmake flags)

I wonder if we could consider adding support for running both wasm32 and wasm64 on wamr64 runtime from the start? Our team is very interested in that scenario as that'd enable us to do a/b testing without having sequential native updates. I'd be happy to help with the work around that.

> WASMMemory stay unchanged
I agree that max page count bigger than UINT32 doesn't make sense; but just out of curiosity, would that break the ABI for AOT? Or what's the reason for not making the change to conform the spec?

TianlongLiang, 2024-01-31:
> I wonder if we could consider adding support for running both wasm32 and wasm64 on wamr64 runtime from the start? Our team is very interested in that scenario as that'd enable us to do a/b testing without having sequential native updates. I'd be happy to help with the work around that.

Yes, I think we can definitely support both from the start, it wouldn't change too much from my original design, mostly some instantiating logic changes I guess. I was merely considering my bandwidth so aiming for simplicity for testing. With your team's help, I think we can definitely support both from the start. 

> I agree that max page count bigger than UINT32 doesn't make sense; but just out of curiosity, would that break the ABI for AOT? Or what's the reason for not making the change to conform the spec? If it's due to ABI change, I think it's ok since we're breaking the ABI with this feature anyway.

You are right that we are breaking AOT ABI anyway, but breaking ABI is not my main concern. I was thinking since we won't be using more pages, changing the type would change more code logic for no actual benefits but also take up more space, making the whole changes more complex.

I just realized that my statement can cause some confusion here. What I truly mean is that:
- In the loader, we can conform to the spec, loading any uint64 values that the spec allows, but the wasm loader can set a special value(like UINT32MAX - 1) value to indicate it is larger than the allowed value(I was thinking maybe we can set a reasonable allowed value like 32GB or something?).

- As long as you are not instantiating it, you are fine. But if you do, then in the wasm instantiating stage or in the wamrc compiling stage we will test if the special value of page count is set and correspondingly throw an exception.

Of course, I could miss something, feel free to point it out and we can discuss other alternatives together 😄

loganek, 2024-01-31:
> So we can change those types to:
>
> typedef mem_offset_t = uint32/uint64; based on CMake flag
> uintptr_t;
> uint64;

I'd strongly discourage us from using `uintptr_t` - this is used for representing the host address, and using that for linear memory address is not good for the following reasons:
* it might be not big enough (if we ever, for whatever reason, decide to run wasm64 on 32bit systems)
* it might be unnecessarily big (e.g. in case of wasm32 running on x86_64)
* it's confusing (if I see `uintptr_t`, I immediately think of the host virtual memory, not linear memory)

I think the first option:
```
#ifdef WASM64
typedef mem_offset_t uint64;
#else
typedef mem_offset_t uint32;
#endif
```
looks best. I'm not 100% sure though whether `offset` is the best name to represent the address; I'd assume the offset can be negative, but if we want to use it to represent an absolute address, then perhaps we could use `mem_address_t` or `linear_mem_ptr_t` etc.

loganek, 2024-01-31:
Also, we'll be having a discussion next week in the WASI SG about memory64 support in WASI-libc/WASI SDK: https://github.com/WebAssembly/meetings/pull/1479

TianlongLiang, 2024-02-01:
> looks best. I'm not 100% sure though whether offset is the best name to represent the address; I'd assume the offset can be negative, but if we want to use it to represent an absolute address, then perhaps we could use mem_address_t or linear_mem_ptr_t etc.

I think `linear_mem_ptr_t` is a solid choice

wenyongh, 2024-02-02:
Hi, for the linear memory related APIs exposed (and its internal APIs), I think there may be several questions to discuss: what is type of related linear memory offset arguments/results in these APIs? Whether to define mem_offset_t or linear_mem_ptr_t type? And whether to use cmake flag to control it?

Note that developer may directly include wasm_export.h and use wamr static/shared lib in his project, in which the cmake flag in WAMR won't take effect in wasm_export.h, so we should not use internal macro to control API definitions in wasm_export.h.

So my opinion is that there may be two options:
- One is to use `uint64` as the type of linear memory offset for both 32-bit target and 64-bit target, and all related APIs are changed to uint64 offset. And no need to introduce mem_offset_t or linear_mem_ptr_t.
- The other is to use `uint32` in 32-bit target and `uint64` for 64-bit target, and use macro `#if UINTPTR_MAX == UINT64_MAX (or UINT32_MAX)` to auto-detect the bit width of the target, and define mem_offset_t or something else accordingly. This supposes that 32-bit target only supports memory32 and memory64 is unsupported. And note that in 64-bit target, the offset is uint64 no matter the linear memory is memory32 or memory64.

I think the first one simplifies the implementation and reduces the maintain effort, but it may slightly impact the footprint and performance. Since there are too many features now and the source code is already very complex, it is my personal preference.

BTW, we had better refactor the current code to change API definitions and data structure definitions (memory instance, module instance, etc.) and resolve historical issues (e.g. UINT32_MAX linear memory size limitation), to make the AOT ABI well defined before we start to implement memory64, since we may release 2.0.0 after GC is merged and before memory64 is implemented.

wenyongh, 2024-03-12:
https://github.com/bytecodealliance/wasm-micro-runtime/commit/0ee5ffce8573a0e41f5a33dce562f541e98eb28a

wenyongh, 2024-03-29:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3240
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3260
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3266

yamt, 2024-05-07:
do you have any plans on table64? https://github.com/WebAssembly/memory64/issues/51

TianlongLiang, 2024-05-07:
Currently, there is no plan regarding the table64 proposal implementation,  a corresponding new RFC issue probably would be open if anyone were to implement it

trcrsired, 2024-06-16:
They clearly f things up on 64 bit support. I want to duplicate all wasi apis to have a _wasm64 suffix but they seem to even ignore what i am saying. I would like to hear what you guys think.

loganek, 2024-06-29:
Thanks @gajanak , this is a known issue and we were planning to address it as part of the WASI work (overall, memory64 support is in still quite an experimental stage)

TianlongLiang, 2024-07-04:
And one more TODO in the future, not WIP currently, but listed here for reference: [Implement table64 extension](https://github.com/WebAssembly/memory64/issues/51)

fridaymore, 2024-08-27:
Great to see this being worked on -- any insight into when this feature will be ready for use in a development setting and in a production setting?

TianlongLiang, 2024-08-27:
I start to work on table64 and hopefully, it can be ready in a month or so

unvestigate, 2025-02-08:
Hi. Any news regarding this? I am liking WAMR a lot but the issues with strings, as reported by @gajanak above, essentially prevent me from using it in 64-bit mode which may be a dealbreaker for me.

TianlongLiang, 2025-02-10:
Hi, @unvestigate, are you using some C standard library function? As mentioned before, the WASI-p1 does not support memory64, but you can use some libc function through WAMR libc-builtin: you can check this [file](https://github.com/web-devkits/wasm2native/blob/main/core/iwasm/libraries/libc-builtin/libc64_builtin_wrapper.c), if it is sufficient for your usage I will port it to the WAMR repo

unvestigate, 2025-02-10:
I am gonna be honest, this whole WASM world is new to me. What I am trying to do is host a 64-bit wasm module inside a 64-bit host application. The module, written in Zig, has previously been compiled as a dll and is already fairly large (50 000 LoC). I currently compile it as a WASI-enabled module because it doesn't seem to build with the freestanding configuration, but I don't knowingly use any WASI-functionality. Instead I want my host application to provide all system-level functions to the module, eg. printf().

I do however, need to be able to pass strings between the host and the module, which is how I ran into the issue I mentioned. If I try to register a simply logging function with the signature "(*~)" I get an error when loading a module in 64-bit mode. I don't have the error at hand right now, but I can look it up if you want.

unvestigate, 2025-02-24:
Hi @TianlongLiang I kinda feel like this was left hanging. Did you need some info from me or was my use case clear? I am currently investigating if I could use WAMR in 32-bit mode, but full 64-bit support would make it infinitely more useful to me. Cheers!

TianlongLiang, 2025-05-22:
Hi, @unvestigate, sorry for missing your comments, I must have been on vacation or so. I agree that 64-bit support for libc-builtin would be useful, and I will port it soon.

TianlongLiang, 2025-05-22:
Hi, @gajanak, I kept this issue open because the wasi-p1 support for memory64 is still not there. I plan to create the 64-bit version of libc-builtin. Would it be good enough for your use case?