wasm-micro-runtime/wasm-micro-runtime #3103: [wasi-nn] requirement of `load_by_name` implementation

lum1n0us, 2024-01-31:
There is a recently added function, [`load-by-name`](https://github.com/WebAssembly/wasi-nn/blob/main/wit/wasi-nn.wit#L100-L105), in wasi-nn.

```
    /// Load a `graph` by name.
    ///
    /// How the host expects the names to be passed and how it stores the graphs for retrieval via
    /// this function is **implementation-specific**. This allows hosts to choose name schemes that
    /// range from simple to complex (e.g., URLs?) and caching mechanisms of various kinds.
    load-by-name: func(name: string) -> result<graph, error>;
```

Plus, the import name of wasi-nn now is [*wasi_ephemeral_nn*](https://github.com/bytecodealliance/wasi-nn/blob/c50dce92a4051f342a07de8b2cb8ad6bd84ffe46/rust/src/generated.rs#L304)

lum1n0us, 2024-01-31:
@tonibofarull

tonibofarull, 2024-01-31:
I'll start working on it as soon as I have some time.
I'll post a message here to let you know when I start, in case anyone else wants to pick it up first.

ayakoakasaka, 2024-02-04:
@lum1n0us 
`[This interface](https://github.com/WebAssembly/wasi-nn/blob/main/wit/wasi-nn.wit)` is already WIT compliant and uses resources.
Are there any guidelines or directions for that part as WAMR?

lum1n0us, 2024-02-04:
NO. we don't have such guidelines yet. Or, following [witx](https://github.com/WebAssembly/wasi-nn/blob/main/wasi-nn.witx#L67-L70) is another choice.

lum1n0us, 2024-05-28:
@ayakoakasaka I am about to synchronize wamr wasi-nn backends(@ *core/iwasm/libraries/wasi-nn*) to commit(https://github.com/WebAssembly/wasi-nn/commit/c0840c26066fa4dbef12d6b0f3d4762f9d6d43c7). It may need to modify several definitions in *wasi_nn.h* and *wasi_nn_types.h*. Please let me know if there is any concern.