wasm-micro-runtime/wasm-micro-runtime #3119: Protection of executable code

proydakov, 2024-02-02:
Hi, WAMR team.

I have a small C program with memory issue:

```
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv)
{
    char *buf;

    printf("Hello world!\n");

    int32_t* address = (void*)(&printf);
    for(size_t i = 255; i < 285; i++)
    {
        printf("i: %ld address: %p value: %d\n", i, address + i, address[i]);
        address[i] = 0;
    }

    printf("Hello world After Modification!\n");

    buf = malloc(1024);
    if (!buf) {
        printf("malloc buf failed\n");
        return -1;
    }

    printf("buf ptr: %p\n", buf);

    sprintf(buf, "%s", "1234\n");
    printf("buf: %s", buf);

    free(buf);
    return 0;
}
```

Compilation process with WASI_SDK_20:

```
${WASI_SDK_PATH}/bin/clang -O3 -o test.wasm test.c
```

It continues to work as if nothing had happened. The result of the program is incorrect.

```
$ ./iwasm test.wasm
Hello world!
i: 255 address: 0x3fd value: 754974720
i: 256 address: 0x401 value: 538976299
i: 257 address: 0x405 value: 2016434224
i: 258 address: 409 value: 1479552256
i: 259 address: 40d value: 542650411
i: 260 address: 411 value: 808278064
i: 261 address: 415 value: 2016422776
i: 262 address: 419 value: 7876640
i: 263 address: 41d value: 979793250
i: 264 address: 421 value: 7546144
i: 265 address: 425 value: 7233902
i: 266 address: 429 value: 6712937
i: 267 address: 42d value: 1819042157
i: 268 address: 431 value: 1646289775
i: 269 address: 435 value: 1713399413
i: 270 address: 439 value: 1701603681
i: 271 address: 43d value: 1095630948
i: 272 address: 441 value: 1313407054
i: 273 address: 445 value: 3014726
i: 274 address: 449 value: 1819635240
i: 275 address: 44d value: 1207970156
i: 276 address: 451 value: 1869376613
i: 277 address: 455 value: 1919907616
i: 278 address: 459 value: 1092641900
i: 279 address: 45d value: 1919251558
i: 280 address: 461 value: 1685015840
i: 281 address: 465 value: 1667851881
i: 282 address: 469 value: 1869182049
i: 283 address: 46d value: 1207968110
i: 284 address: 471 value: 1869376613

buf ptr: 11520
```

If I build the same code for amd64:

```
clang -O3 -o test test.c
```

I get an immediate stop in the execution process and can track the incorrect stack through gdb.

```
$ ./test
Hello world!
i: 255 address: 0x7f89554a9aec value: 17439
Segmentation fault (core dumped)
```

**Is it possible to create an interpreter execution mode, which would allow you to catch such errors, even at the cost of stopping the entire application during development?**

As I see the memory layout contains `wasm_module_memory`, but I don't know how to get a code size. Also we should use a mmap pages for code. https://bytecodealliance.github.io/wamr.dev/blog/the-wamr-memory-model/

```
int make_memory_executable(void* m, size_t size) {
    if (mprotect(m, size, PROT_READ | PROT_EXEC) == -1) {
        perror("mprotect");
        return -1;
    }
    return 0;
}
```

wenyongh, 2024-02-04:
@proydakov The linear memory is set readable/writable after mapped:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/529fa9dd17bd6bb3159711881cac8def76df7277/core/iwasm/common/wasm_memory.c#L680-L684

If you want to set it executable, please add `| MMAP_PROT_EXEC` to it.

For making your program trap, maybe you can reserve some space at the beginning of the linear memory, set the space inaccessible, e.g. after the above code, add
```C
os_mprotect(new_mem, 4096, MMAP_PROT_NONE);
```
And then add `-Wl,--global-base=4096` for wasi-sdk clang when building the wasm app. I have a try, the "out of bounds memory access" was thrown after that.

yamt, 2024-02-05:
>     int32_t* address = (void*)(&printf);

i guess your intention here is to modify the printf code.
but your test program is not doing that.
wasm uses distinct address spaces for function and data.
`(void*)(&printf)` points to some unrelated data, which happened to have the same address as printf.