wasm-micro-runtime/wasm-micro-runtime #3122: Invalid Read Memory in check_wasi_abi_compatibility

haruki3hhh, 2024-02-03:
## Version
commit b3f728ceb36f9c72047a934436ef41699643ab99 (HEAD -> main, origin/main, origin/HEAD)
Author: Wenyong Huang <wenyong.huang@intel.com>
Date:   Thu Feb 1 12:42:12 2024 +0800

## Compile
```sh
cd wasm-micro-runtime/product-mini/platforms/linux/
cmake -DCMAKE_C_COMPILER="gcc" -DCMAKE_C_FLAGS="-g -fsanitize=address -fno-omit-frame-pointer" -DCMAKE_LINKER_FLAGS="-g -fsanitize=address -fno-omit-frame-pointer" ..
make -j
```

## Reproduce
```sh
./iwasm /root/Ablation/wasm-fuzz/fuzz_out/crashes/id\:000000\,sig\:06\,src\:001224\,op\:python\,pos\:0
```

## ASAN Log
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1546471==ERROR: AddressSanitizer: SEGV on unknown address 0x603800000368 (pc 0x55f3ba9bb762 bp 0x7ffd6d5a03c0 sp 0x7ffd6d5a0260 T0)
==1546471==The signal is caused by a READ memory access.
    #0 0x55f3ba9bb761 in check_wasi_abi_compatibility /root/Ablation/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4276
    #1 0x55f3ba9bb761 in wasm_loader_load /root/Ablation/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4383
    #2 0x55f3ba93171c in main /root/Ablation/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:885
    #3 0x7f74763de082 in __libc_start_main ../csu/libc-start.c:308
    #4 0x55f3ba932b2d in _start (/root/Ablation/wasm-micro-runtime/product-mini/platforms/linux/build_asan/iwasm+0x22b2d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/Ablation/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4276 in check_wasi_abi_compatibility
==1546471==ABORTING
```
## PoC
[PoC](https://github.com/haruki3hhh/fuzzing/blob/main/id%3A000000%2Csig%3A06%2Csrc%3A001224%2Cop%3Apython%2Cpos%3A0)

wenyongh, 2024-02-04:
@haruki3hhh thanks for spotting the issue, it was fixed with #3126.

wenyongh, 2024-03-15:
Close the issue since it had been fixed.

closed 2024-03-15 by wenyongh