wasm-micro-runtime/wasm-micro-runtime #3123: Unexpected Memory OOB Exception ion the fast interpreter mode

erxiaozhou, 2024-02-03:
# Build commands
I compile the code with commit id 529fa9dd17bd6bb3159711881cac8def76df7277.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Release -DWAMR_BUILD_FAST_INTERP=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -Bbuild
```

execute:

```
iwasm --heap-size=0  <tc_name>
```
# Cases

[fast_nterpreter_unexpected_MEMOOB.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14152217/fast_nterpreter_unexpected_MEMOOB.zip)

## Code in WAT
```
(module
  (type (;0;) (func (param i32 i64) (result f64)))
  (type (;1;) (func (result f64)))
  (func (;0;) (type 1) (result f64)
    (local i32 f32 i64 f64)
    i32.const 0
    i64.const -21368906167275783
    i32.const 40
    if  ;; label = @1
    else
      f64.const 0x1.8p+1 (;=3;)
      return
      drop
      drop
      i32.const 1
      local.set 0
    end
    block (param i32 i64) (result f64)  ;; label = @1
      global.get 0
      i64.add
      global.set 0
      i64.const 0
      i64.store16 offset=10318 align=1
      local.get 0
      f64.const 0x1.8p+1 (;=3;)
      br 1 (;@0;)
    end)
  (memory (;0;) 1 5)
  (global (;0;) (mut i64) (i64.const 0))
  (export "to_test" (func 0)))

```

# Actual output:

```
Exception: out of bounds memory access
```

# Expected output:
```
3:f64
```

wenyongh, 2024-02-06:
@erxiaozhou the issue was fixed with #3135, please try again. Thanks for reporting the issue.

wenyongh, 2024-03-15:
Close the issue since it had been fixed.

closed 2024-03-15 by wenyongh