wasm-micro-runtime/wasm-micro-runtime #3124: thread_spawn_wrapper  CRASH, exec_env is NULL

JakeOrange, 2024-02-04:
When I debug 
[wasi_threads_exit_main_block.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14155395/wasi_threads_exit_main_block.wasm.zip) on the MacOS and Android， thread_spawn_wrapper  CRASH, exec_env is NULL。 I had setting `WASM_ENABLE_INTERP=1, WASM_ENABLE_FAST_INTERP=1, WASM_ENABLE_LIBC_WASI=1, WASM_ENABLE_LIBC_BUILTIN=1, WASM_ENABLE_REF_TYPES, WASM_ENABLE_LIB_WASI_THREADS=1, WASM_ENABLE_BULK_MEMORY=1, WASM_ENABLE_SHARED_MEMORY=1, WASM_ENABLE_THREAD_MGR=1, CLUSTER_MAX_THREAD_NUM=12, BH_FREE=wasm_runtime_free, BH_MALLOC=wasm_runtime_malloc`.

yamt, 2024-02-05:
> When I called `_start` in the [wasi_threads_exit_main_block.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14155395/wasi_threads_exit_main_block.wasm.zip) on the MacOS and Android， thread_spawn_wrapper CRASH, exec_env is NULL。 I had setting `WASM_ENABLE_INTERP=1, WASM_ENABLE_FAST_INTERP=1, WASM_ENABLE_LIBC_WASI=1, WASM_ENABLE_LIBC_BUILTIN=1, WASM_ENABLE_REF_TYPES, WASM_ENABLE_LIB_WASI_THREADS=1, WASM_ENABLE_BULK_MEMORY=1, WASM_ENABLE_SHARED_MEMORY=1, WASM_ENABLE_THREAD_MGR=1, CLUSTER_MAX_THREAD_NUM=12, BH_FREE=wasm_runtime_free, BH_MALLOC=wasm_runtime_malloc`.

i couldn't reproduce it with iwasm on macOS x86-64.

JakeOrange, 2024-02-05:
> i couldn't reproduce it with iwasm on macOS x86-64.

After I instantiated the module, I only call the `_start` function. No any host code. I have no idea. 
Do you have any suggestions for checking?

[image]
[image]

lum1n0us, 2024-02-06:
Also couldn't reproduce it with `iwasm -v=5 -f _start wasi_threads_exit_main_block.wasm`.

Here are some debugging BKMs.

- use `iwasm -f _start` or `iwasm` to reproduce.
- `wasm_runtime_invoke_native()` prepares parameters and stores in `argv1`. please check the value of `argv1`, `fps`, `ints`, `stacks`, `n_ints`, `n_stacks`. especially, values in `ints`. if using lldb, you might want to `memory read -f x --size 4 --count 8 ints`
- after `wasm_runtime_invoke_native()`, goes to *invokeNative_XXX.s*, `XXX` should be *em64* or *em64_simd* depends on if enabling `WAMR_BUILD_SIMD` or not.

JakeOrange, 2024-02-06:
SIMD open cause the issue. Thank you very much!

closed 2024-02-06 by JakeOrange