wasm-micro-runtime/wasm-micro-runtime #3130: NULL Pointer Dereference in block_type_get_result_types

haruki3hhh, 2024-02-05:
### Version
commit 06df58f20eeb46351b0ca7ff46d21eb541d2784a (HEAD -> main, origin/main, origin/HEAD)

### Compile
```sh
cd wasm-micro-runtime/product-mini/platforms/linux/
cmake -DCMAKE_C_COMPILER="gcc" -DCMAKE_C_FLAGS="-g -fsanitize=address -fno-omit-frame-pointer" -DCMAKE_LINKER_FLAGS="-g -fsanitize=address -fno-omit-frame-pointer" ..
make -j
```
### ASAN Log
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==866714==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x565157bc383f bp 0x7ffe6cfb60d0 sp 0x7ffe6cfb5f90 T0)
==866714==The signal is caused by a READ memory access.
==866714==Hint: address points to the zero page.
    #0 0x565157bc383e in block_type_get_result_types /root/wasm-micro-runtime/core/iwasm/interpreter/wasm.h:875
    #1 0x565157bc383e in block_type_get_result_types /root/wasm-micro-runtime/core/iwasm/interpreter/wasm.h:863
    #2 0x565157bc383e in wasm_loader_check_br /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7092
    #3 0x565157bc383e in check_branch_block /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:7143
    #4 0x565157bd0a73 in wasm_loader_prepare_bytecode /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:8127
    #5 0x565157bd0a73 in load_from_sections /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:3852
    #6 0x565157bdf654 in load /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4215
    #7 0x565157bdf654 in wasm_loader_load /root/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:4391
    #8 0x565157b5571c in main /root/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:885
    #9 0x7f54f8c2a082 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #10 0x565157b56b4d in _start (/root/wasm-micro-runtime/product-mini/platforms/linux/build_asan/iwasm+0x22b4d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/wasm-micro-runtime/core/iwasm/interpreter/wasm.h:875 in block_type_get_result_types
==866714==ABORTING

```

### Reproduce
```sh
./iwasm /root/wasm-fuzz-instruction/fuzz_out/crashes/id\:000000\,sig\:06\,src\:028368\,op\:python\,pos\:0
```

### PoC
[PoC](https://github.com/haruki3hhh/fuzzing/blob/main/id%3A000000%2Csig%3A06%2Csrc%3A028368%2Cop%3Apython%2Cpos%3A0)

wenyongh, 2024-02-05:
@haruki3hhh Thanks for reporting this issue! I submitted PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/3133 to fix it, please try again.

wenyongh, 2024-03-15:
Close this issue since it was resolved.

closed 2024-03-15 by wenyongh