wasm-micro-runtime/wasm-micro-runtime #3138: "wamrc --xip" triggers an assertion failure in llvm

yamt, 2024-02-06:
`wamrc --target=xtensa --xip -o aot ~/git/garbage/wasm/hello/hello.wasm` triggered
the following assertion failure.

```
Assertion failed: (isa<To>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file Casting.h, line 579.
Process 57991 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = hit program assert
    frame #4: 0x000000010074e395 wamrc`decltype(Val=0x00006000026165a0) llvm::cast<llvm::Function, llvm::Value>(llvm::Value*) at Casting.h:579:3
   576 
   577  template <typename To, typename From>
   578  [[nodiscard]] inline decltype(auto) cast(From *Val) {
-> 579    assert(isa<To>(Val) && "cast<Ty>() argument of incompatible type!");
   580    return CastInfo<To, From *>::doCast(Val);
   581  }
   582 
Target 0: (wamrc) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = hit program assert
    frame #0: 0x00007ff80bd89ffe libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007ff80bdc01ff libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007ff80bd0bd14 libsystem_c.dylib`abort + 123
    frame #3: 0x00007ff80bd0b0bb libsystem_c.dylib`__assert_rtn + 314
  * frame #4: 0x000000010074e395 wamrc`decltype(Val=0x00006000026165a0) llvm::cast<llvm::Function, llvm::Value>(llvm::Value*) at Casting.h:579:3
    frame #5: 0x000000010071c48d wamrc`llvm::Function* llvm::unwrap<llvm::Function>(P=0x00006000026165a0) at Value.h:1034:1
    frame #6: 0x000000010071f615 wamrc`::LLVMGetFunctionCallConv(Fn=0x00006000026165a0) at Core.cpp:2395:10
    frame #7: 0x0000000100099f18 wamrc`aot_compile_op_call(comp_ctx=0x000000011702a800, func_ctx=0x0000600003b001c0, func_idx=5, tail_call=false) at aot_emit_function.c:1093:47
    frame #8: 0x000000010001998c wamrc`aot_compile_func(comp_ctx=0x000000011702a800, func_index=1) at aot_compiler.c:352:22
    frame #9: 0x0000000100017595 wamrc`aot_compile_wasm(comp_ctx=0x000000011702a800) at aot_compiler.c:2642:14
    frame #10: 0x00000001000082a7 wamrc`main(argc=1, argv=0x00007ff7bfeff5d8) at main.c:664:10
    frame #11: 0x000000011548552e dyld`start + 462
(lldb) 
```

`func` here seems like a return value of `aot_get_func_from_table`.

```
(lldb) p *Val
(llvm::Value) $5 = {
  VTy = 0x0000000117046e00
  UseList = 0x000060000301b980
  SubclassID = '<'
  HasValueHandle = '\0'
  SubclassOptionalData = '\0'
  SubclassData = 4
  NumUserOperands = 1
  IsUsedByMD = 0
  HasName = 1
  HasMetadata = 0
  HasHungOffUses = 0
  HasDescriptor = 0
}
(lldb) p *Val->VTy
(llvm::Type) $6 = {
  Context = 0x0000600000004360
  ID = PointerTyID
  SubclassData = 0
  NumContainedTys = 0
  ContainedTys = nullptr
}
(lldb) 
```

yamt, 2024-02-06:
@wenyongh 
do you remember in what situation this LLVMSetInstructionCallConv was necessary?

wenyongh, 2024-02-06:
I can't remember why it was added, maybe it just ensures the callee's call convention is same as the caller, but it seems that it is unnecessary? Maybe you can try to remove it?

yamt, 2024-02-07:
fixed with https://github.com/bytecodealliance/wasm-micro-runtime/pull/3140

closed 2024-02-07 by yamt