wasm-micro-runtime/wasm-micro-runtime #3139: samples/multi-thread build failed

xianjimli, 2024-02-06:
reproduce steps:

```
cd wasm-micro-runtime/samples/multi-thread
cmake .
make
```

output log

```
[ 90%] Building C object wasm-apps/CMakeFiles/test.wasm.dir/main.c.o
[ 92%] Linking C executable test.wasm
[wasm-validator error in function 1] unexpected false: Atomic operations require threads [--enable-threads], on 
(i32.atomic.rmw.cmpxchg
 (i32.const 1272)
 (i32.const 0)
 (i32.const 1)
)
[wasm-validator error in function 1] unexpected false: Bulk memory operations require bulk memory [--enable-bulk-memory], on 
(memory.init $0
 (i32.const 1024)
 (i32.const 0)
 (i32.const 234)
)
[wasm-validator error in function 1] unexpected false: Bulk memory operations require bulk memory [--enable-bulk-memory], on 
(memory.fill
 (i32.const 1260)
 (i32.const 0)
 (i32.const 12)
)
[wasm-validator error in function 1] unexpected false: Atomic operations require threads [--enable-threads], on 
(i32.atomic.store
 (i32.const 1272)
 (i32.const 2)
)
[wasm-validator error in function 1] unexpected false: Atomic operations require threads [--enable-threads], on 
(memory.atomic.notify
 (i32.const 1272)
 (i32.const -1)
)
[wasm-validator error in function 1] unexpected false: Atomic operations require threads [--enable-threads], on 
(memory.atomic.wait32
 (i32.const 1272)
 (i32.const 1)
 (i64.const -1)
)
[wasm-validator error in function 1] unexpected false: Bulk memory operations require bulk memory [--enable-bulk-memory], on 
(data.drop $0)
[wasm-validator error in module] unexpected false: shared memory requires threads [--enable-threads], on 
memory
[wasm-validator error in module] unexpected false: nonzero segment flags require bulk memory [--enable-bulk-memory], on 
Fatal: error validating input
clang-16: error: linker command failed with exit code
```

lum1n0us, 2024-02-06:
May I ask about the platform and cpu architecture.

xianjimli, 2024-02-06:
macos aarch64

lum1n0us, 2024-02-06:
macos 1x?

lum1n0us, 2024-02-06:
FYI: 

WAMR has a per-PR/per-Merage validation on macOS 14.2.1 + arm64.

https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/7794447983/job/21255864582

lum1n0us, 2024-02-06:
may I ask for the output from `cmake .`? It should be something:

```
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled

...
```

xianjimli, 2024-02-06:
> macos 1x?
yes

xianjimli, 2024-02-06:
cmake . output:

```
-- Build Configurations:
     Build as target AARCH64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter enabled
     Multiple modules disabled
     Bulk memory feature enabled
     Shared memory enabled
     Thread manager enabled
     Lib pthread enabled
     Lib pthread semaphore enabled
     Wakeup of blocking operations enabled
     Reference types disabled
     Quick AOT/JIT entries enabled
CMake Deprecation Warning at wasm-apps/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (0.0s)
-- Generating done (0.0s)

```

lum1n0us, 2024-02-06:
any information about the compiler? likes: 

```
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0

```

xianjimli, 2024-02-06:
```
/Library/Developer/CommandLineTools/usr/bin/cc -DBH_FREE=wasm_runtime_free -DBH_MALLOC=wasm_runtime_malloc -DBH_PLATFORM_DARWIN -DBUILD_TARGET=\"AARCH64\" -DBUILD_TARGET_AARCH64 -DWASM_DISABLE_HW_BOUND_CHECK=0 -DWASM_DISABLE_STACK_HW_BOUND_CHECK=0 -DWASM_DISABLE_WAKEUP_BLOCKING_OP=0 -DWASM_ENABLE_AOT=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWASM_ENABLE_FAST_INTERP=1 -DWASM_ENABLE_INTERP=1 -DWASM_ENABLE_LIBC_BUILTIN=1 -DWASM_ENABLE_LIB_PTHREAD=1 -DWASM_ENABLE_LIB_PTHREAD_SEMAPHORE=1 -DWASM_ENABLE_MINI_LOADER=0 -DWASM_ENABLE_MULTI_MODULE=0 -DWASM_ENABLE_QUICK_AOT_ENTRY=1 -DWASM_ENABLE_SHARED_MEMORY=1 -DWASM_ENABLE_THREAD_MGR=1 -DWASM_GLOBAL_HEAP_SIZE=10485760 -DWASM_HAVE_MREMAP=0 -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/interpreter -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/aot -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-builtin -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/libraries/lib-pthread -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/libraries/thread-mgr -I/Users/jim/work/lab/wasm/wasm-micro-runtime/samples/multi-thread/../../core/iwasm/include -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/darwin -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/darwin/../include -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/mem-alloc -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/common -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/utils -I/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/utils/uncommon -Wall -Wextra -Wformat -Wformat-security -std=gnu99 -ffunction-sections -fdata-sections                                          -Wall -Wno-unused-parameter -Wno-pedantic -fPIC -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk -fPIE -MD -MT CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/product-mini/platforms/linux/main.c.o -MF CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/product-mini/platforms/linux/main.c.o.d -o CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/product-mini/platforms/linux/main.c.o -c /Users/jim/work/lab/wasm/wasm-micro-runtime/product-mini/platforms/linux/main.c
```

jim@lixianjingdeMacBook-Pro multi-thread % /Library/Developer/CommandLineTools/usr/bin/cc --version

```
Apple clang version 15.0.0 (clang-1500.1.0.2.5)
Target: arm64-apple-darwin23.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

lum1n0us, 2024-02-06:
it is for *product-mini/platforms/linux/main.c*. I hope to know the compiler information for *wasm-apps/*"

xianjimli, 2024-02-06:
```
jim@lixianjingdeMacBook-Pro wasm-rt % /opt/wasi-sdk/bin/clang --version
clang version 16.0.0 (https://github.com/llvm/llvm-project 08d094a0e457360ad8b94b017d2dc277e697ca76)
Target: wasm32-unknown-wasi
Thread model: posix
InstalledDir: /opt/wasi-sdk/bin
```

lum1n0us, 2024-02-06:
... 🤔 ...

There is a *link.txt* under, maybe, *wasm-apps/CMakeFiles/test.wasm.dir/*. Would you copy and paste its content here?

xianjimli, 2024-02-06:
./wasm-apps/CMakeFiles/test.wasm.dir/link.txt

```
/opt/wasi-sdk/bin/clang --target=wasm32 --sysroot=/Users/jim/work/lab/wasm/wasm-micro-runtime/samples/multi-thread/wasm-apps/../../../wamr-sdk/app/libc-builtin-sysroot -nostdlib -pthread -Qunused-arguments -z stack-size=32768 -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk -Wl,--shared-memory,--max-memory=131072,                -Wl,--no-entry,--strip-all,                             -Wl,--export=__heap_base,--export=__data_end            -Wl,--export=__wasm_call_ctors                          -Wl,--export=main -Wl,--export=__main_argc_argv         -Wl,--allow-undefined  -fPIC CMakeFiles/test.wasm.dir/main.c.o -o test.wasm 
```

lum1n0us, 2024-02-06:
please give it a try, like:

```
cmake -DCMAKE_BUILD_TYPE=Debug .
make 
```

lum1n0us, 2024-02-06:
- please take a look at your wasi-sdk version. better use the [latest one](https://github.com/WebAssembly/wasi-sdk/releases/tag/wasi-sdk-21).
- if there is a wasm-opt in *${wasi-sdk}/bin/*, please let me know and remove it. 

after above steps, please reproduce.

xianjimli, 2024-02-06:
wasi-sdk is built with lastest source:

```
jim@lixianjingdeMacBook-Pro multi-thread % /opt/wasi-sdk/bin/clang --version
clang version 17.0.6 (https://github.com/llvm/llvm-project 6009708b4367171ccdbf4b5905cb6a803753fe18)
Target: wasm32-unknown-wasi
Thread model: posix
InstalledDir: /opt/wasi-sdk/bin
```

rebuild

```
jim@lixianjingdeMacBook-Pro samples % cd multi-thread 
jim@lixianjingdeMacBook-Pro multi-thread % cmake -DCMAKE_BUILD_TYPE=Debug .
make

```

output log
```
-- The C compiler identification is AppleClang 15.0.0.15000100
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build Configurations:
     Build as target AARCH64
     CMAKE_BUILD_TYPE Debug
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT disabled
     WAMR LLVM ORC JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter enabled
     Multiple modules disabled
     Bulk memory feature enabled
     Shared memory enabled
     Thread manager enabled
     Lib pthread enabled
     Lib pthread semaphore enabled
     Wakeup of blocking operations enabled
     Reference types disabled
     Quick AOT/JIT entries enabled
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Looking for mremap
-- Looking for mremap - not found
CMake Deprecation Warning at wasm-apps/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (0.7s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/jim/work/lab/wasm/wasm-micro-runtime/samples/multi-thread
[  1%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/darwin/platform_init.c.o
[  3%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/common/memory/mremap.c.o
[  5%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_blocking_op.c.o
[  7%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_malloc.c.o
[... 32 lines cut ...]
[ 71%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/aot/aot_loader.c.o
[ 73%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/aot/aot_runtime.c.o
[ 75%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/aot/arch/aot_reloc_aarch64.c.o
[ 76%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/libraries/lib-pthread/lib_pthread_wrapper.c.o
[ 78%] Building C object CMakeFiles/vmlib.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/iwasm/libraries/thread-mgr/thread_manager.c.o
[ 80%] Linking C static library libvmlib.a
[ 80%] Built target vmlib
[ 82%] Building C object CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/product-mini/platforms/linux/main.c.o
[ 84%] Building C object CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/utils/uncommon/bh_getopt.c.o
[ 86%] Building C object CMakeFiles/iwasm.dir/Users/jim/work/lab/wasm/wasm-micro-runtime/core/shared/utils/uncommon/bh_read_file.c.o
[ 88%] Linking C executable iwasm
[ 88%] Built target iwasm
[ 90%] Building C object wasm-apps/CMakeFiles/test.wasm.dir/main.c.o
[ 92%] Linking C executable test.wasm
[ 92%] Built target test.wasm
[ 94%] Building C object wasm-apps/CMakeFiles/main_thread_exception.wasm.dir/main_thread_exception.c.o
[ 96%] Linking C executable main_thread_exception.wasm
[ 96%] Built target main_thread_exception.wasm
[ 98%] Building C object wasm-apps/CMakeFiles/main_global_atomic.wasm.dir/main_global_atomic.c.o
[100%] Linking C executable main_global_atomic.wasm
[100%] Built target main_global_atomic.wasm
```
> conclusion: lastest wasi-sdk works.

Thank you

eloparco, 2024-02-06:
@lum1n0us Actually I was able to reproduce on my M1 Mac. Never noticed that before because usually I build in debug mode (and that works), but if I try release mode I get the same error reported in this issue.

lum1n0us, 2024-02-07:
The issue is caused by wasm-opt of binaryen (which I guess is in `$PATH`) who can't share clang's command options for now. https://github.com/llvm/llvm-project/issues/55781 could give more details.

The workaround is to remove wasm-opt from `$PATH`.