wasm-micro-runtime/wasm-micro-runtime #3143: pthread_library.md does not work on macpro M1

xianjimli, 2024-02-06:
main.c

```c
#include <stdint.h>
#include <stdlib.h>
#include <stdio.h>
#include <pthread.h>

void *thread_routine(void *arg)
{
    printf("Enter thread\n");
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char** argv)
{
    pthread_t tid;

    if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        printf("Failed to create thread\n");
    }   

    if (0 != pthread_join(tid, NULL)) {
        printf("Failed to join thread %d.\n", tid);
    }   

    printf("Exit\n");

    return 0;
}
```

build.sh

```
WAMR_ROOT="/Users/jim/work/lab/wasm/wasm-micro-runtime/"

/opt/wasi-sdk/bin/clang  --target=wasm32        \   
    --sysroot=${WAMR_ROOT}/wamr-sdk/app/libc-builtin-sysroot   \   
    -O3 -pthread -nostdlib -z stack-size=32768      \   
    -Wl,--shared-memory             \   
    -Wl,--initial-memory=131072,--max-memory=131072 \
    -Wl,--allow-undefined-file=${WAMR_ROOT}/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt \
    -Wl,--no-entry -Wl,--export=main                \   
    -Wl,--export=__heap_base,--export=__data_end    \   
    -Wl,--export=__wasm_call_ctors  \
    main.c -o test.wasm
```

run

```
./build.sh
```

output
```
wasm-ld: error: symbol exported via --export not found: main
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

lum1n0us, 2024-03-20:
Thanks for reporting. It seems clang will remove all unused code, in this case `main()` and `thread_routine()`, and wasm-ld can't find it when arguments requires to `--export=main`.

One solution is `__attribute__`:

``` c
__attribute__((export_name("main")))
int main(int argc, char **argv) {
```

xianjimli, 2024-03-21:
ok, thanks