wasm-micro-runtime/wasm-micro-runtime #3147: Android mult-thread pthread_exit crash

JakeOrange, 2024-02-06:
I am debugging the case[hello_world.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14181188/hello_world.wasm.zip) on **Android**.
I instantiated the wasm module. I called the function `_start`. It will create 10 threads. 
After `wasm_func_call` executed successfully,`pthread_exit` crashed.  

[image]

lum1n0us, 2024-02-08:
what is the value of *maximum number of threads*?

JakeOrange, 2024-02-08:
12

JakeOrange, 2024-02-08:
I run it ok on MacOS and iOS.

lum1n0us, 2024-02-18:
In that case, I would suggest setting `max_thread_num` of `RuntimeInitArgs` to a larger number, like 12 or 15, then using `wasm_runtime_full_init()` to start everything.

JakeOrange, 2024-02-19:
> In that case, I would suggest setting `max_thread_num` of `RuntimeInitArgs` to a larger number, like 12 or 15, then using `wasm_runtime_full_init()` to start everything.

I had set `CLUSTER_MAX_THREAD_NUM` to 12. And MacOS and iOS is rightly. Are there other points?

lum1n0us, 2024-02-19:
*iwasm* works well with the *hello_world.wasm* on my local environment. Could u try it, with *iwasm*, on MacOS?

```
$ ./iwasm --max-threads=20 ./hello_world.wasm
 in thread 4
 in thread 7
 in thread 6
 in thread 2
 in thread 3
 in thread 9
 in thread 5
 in thread 0
 in thread 8
 in thread 1
```

JakeOrange, 2024-02-19:
> _iwasm_ works well with the _hello_world.wasm_ on my local environment. Could u try it, with _iwasm_, on MacOS?
> 
> ```
> $ ./iwasm --max-threads=20 ./hello_world.wasm
>  in thread 4
>  in thread 7
>  in thread 6
>  in thread 2
>  in thread 3
>  in thread 9
>  in thread 5
>  in thread 0
>  in thread 8
>  in thread 1
> ```

> I had set `CLUSTER_MAX_THREAD_NUM` to 12. **And MacOS and iOS is rightly.** Are there other points?

I had try it on MacOS, it is OK. It only happen on Android. And it must happen.

lum1n0us, 2024-02-19:
> I had try it on MacOS, it is OK. It only happen on Android. And it must happen.

When reproduce it on Android, using *iwasm* or your local, self compiled, wamr library?

JakeOrange, 2024-02-20:
> > I had try it on MacOS, it is OK. It only happen on Android. And it must happen.
> 
> When reproduce it on Android, using _iwasm_ or your local, self compiled, wamr library?

Compile locally to `.so` with `cmake`

lum1n0us, 2024-02-21:
Since I can't reproduce it locally, here are some BKMs of the team, hope useful:

- Don't mix using `WAMR_BUILD_LIB_PTHREAD` and `WAMR_BUILD_LIB_PTHREAD` together
- With the help of valgrind and sanitizers(`WAMR_BUILD_SANITIZER`)
- Shrink the problem by break something. For example:
  - try variant thread numbers
  - try variant compilation options, like BOUND_CHECK
  - comment `wasm_deinstantiate()`
  - comment `wasm_unload()`
  - ...

JakeOrange, 2024-02-21:
> Since I can't reproduce it locally, here are some BKMs of the team, hope useful:

Get it. Thanks!