wasm-micro-runtime/wasm-micro-runtime #3151: ref.as_non_null cause random result

peter-jerry-ye, 2024-02-07:
The following wat file will create random output 

```wasm
(module
 (data $string_data "asdf")
 (func $print (import "spectest" "print_i32") (param $i i32)) ;; prints an integer, provided by a native lib
 (memory $memory (export "memory") 1)
 (type $string (array (mut i8)))
 (type $var (struct (field (ref null $string))))
 (func $test
  (param $param (ref $var))
  (local.get $param) (struct.get $var 0) 
  (ref.as_non_null) ;; this causes random output; uncomment it and the output will be stable
  (array.len) (call $print))
 (func $init
  (local $str (ref $string))
  (array.new_data $string $string_data (i32.const 0) (i32.const 4)) (local.set $str) ;; create a string that length = 4
  (struct.new $var (local.get $str)) (call $test))
 (export "_start" (func $init)))
```

Remove the `ref.as_non_null` then the output will constantly be 4.

This occurs
- with the latest commit (7cac0531adf7bd4327fa1854fe3481c7887c1799)
- ran with `iwasm` built with `cmake -DWAMR_BUILD_GC=1 ..`
- The environment is MacOS M1 Sonoma 14.2

peter-jerry-ye, 2024-02-08:
I simplified a bit and got unhandled SIGSEGV with
```wasm
(module
 (data $string_data "asdf")
 (func $print (import "spectest" "print_i32") (param $i i32))
 (memory $memory (export "memory") 1)
 (type $string (array (mut i8)))
 (func $init
  (local $str (ref null $string))
  (array.new_data $string $string_data (i32.const 0) (i32.const 4)) (local.tee $str)
  (ref.as_non_null) 
  (array.len) (call $print))
 (export "_start" (func $init)))
```

wenyongh, 2024-02-09:
@peter-jerry-ye thanks for reporting the issue! Please try #3156.

peter-jerry-ye, 2024-02-09:
@wenyongh Thank you for the quick response. I confirm it fixes the problem.

closed 2024-02-19 by peter-jerry-ye