wasm-micro-runtime/wasm-micro-runtime #3159: AOT running mode

proydakov, 2024-02-13:
Hello, WAMR team.

Is it possible to detect that we are starting in AOT mode?

```
/* Running mode of runtime and module instance*/
typedef enum RunningMode {
    Mode_Interp = 1,
    Mode_Fast_JIT,
    Mode_LLVM_JIT,
    Mode_Multi_Tier_JIT,
} RunningMode;

WASM_RUNTIME_API_EXTERN RunningMode
wasm_runtime_get_running_mode(wasm_module_inst_t module_inst);
```

For common .wasm input file I get:

```
'Mode_Interp' // Mode_Interp = 1,
```

For .aot input file I see:

```
'0' // #define Mode_Default 0
```

Best regard, Proydakov Evgeny.

TianlongLiang, 2024-02-19:
Hello, @proydakov, if you are loading the AOT file, you must start in AOT mode, and that is the only way to use AOT mode. There is no export API to determine whether a `wasm_module_inst_t module_inst` is an AOT module instance or not. However, you can use get_package_type to identify whether it's a wasm file or an AOT file using:

```C
/* read wasm/aot file into a memory buffer */
buffer = bh_read_file_to_buffer("../wasm-apps/test.file", &size);
/* whether it's wasm file */
if (get_package_type(buffer, size) == Wasm_Module_Bytecode){
...
}
/* whether it's aot file */
if (get_package_type(buffer, size) == Wasm_Module_AoT){
...
}
```

proydakov, 2024-03-04:
I see:

```
/* Package Type */
typedef enum {
    Wasm_Module_Bytecode = 0,
    Wasm_Module_AoT,
    Package_Type_Unknown = 0xFFFF
} package_type_t;
```

Many Thx.

closed 2024-03-04 by proydakov