wasm-micro-runtime/wasm-micro-runtime #3163: AOT debugging setup doesn't work

g0djan, 2024-02-15:
Hi, I followed https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/source_debugging_aot.md

After building `wamrc` but it fails with coredump(which isn't configured on this ec2) when I try to build AOT files

```
ubuntu@ip-172-31-42-211:~/wasm-micro-runtime$ ./wamr-compiler/build/wamrc --disable-simd --size-level=1 -o helloworld.aot helloworld/target/wasm32-wasi/debug/helloworld.wasm
Create AoT compiler with:
  target:        x86_64
  target cpu:    cascadelake
  target triple: x86_64-unknown-linux-gnu
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
wamrc: /home/ubuntu/wasm-micro-runtime/core/deps/llvm/build/include/llvm/Support/Casting.h:109: static bool llvm::isa_impl_cl<To, const From*>::doit(const From*) [with To = llvm::DIType; From = llvm::Metadata]: Assertion `Val && "isa<> used on a null pointer"' failed.
Aborted (core dumped)
```

If I build `wamrc` without passing `-DWAMR_BUILD_DEBUG_AOT=1` everything works

yamt, 2024-02-29:
is your helloworld.wasm in rust?

i saw a similar symptom with C++ module. https://github.com/bytecodealliance/wasm-micro-runtime/issues/3187

yamt, 2024-02-29:
for rust modules, it seems that, in `lldb_function_to_function_dbi`, LLVMCountParams(func_ctx->func) is sometimes smaller than 1+variable_list.GetSize() for some reasons.
i don't know how it happens.

g0djan, 2024-03-09:
Yes it was Rust helloworld

TianlongLiang, 2024-04-03:
It's because of the same reason mentioned [here](https://github.com/bytecodealliance/wasm-micro-runtime/issues/3187#issuecomment-2011318141), currently exploiting a solution. Will also update here if something new