wasm-micro-runtime/wasm-micro-runtime #3164: AOT compiler `wamrc` doesn't support compiling to arm64 bit

penneryu, 2024-02-16:
penner@xxx build % ./wamrc --target=aarch64 -o wasm.aot wasm.wasm 
[18:08:57:013 - 1DEF6D000]: warning: a module with WASI apis should be either a command or a reactor
Create AoT compiler with:
  target:        aarch64v8
  target cpu:    
  target triple: aarch64v8-pc-linux-gnu
  cpu features:  
  opt level:     3
  size level:    1
  output format: AoT file
LLVM ERROR: Only small, tiny and large code models are allowed on AArch64
zsh: abort      ./wamrc --target=aarch64 -o wasm.aot wasm.wasm

Can I modify the source code in aot_llvm_extra2.cpp

TianlongLiang, 2024-02-19:
Hi, can you tell me which platform and which commit you are using? 
I tried it on my x86 Ubuntu20 with latest commit, it's fine:
```shell
./wamrc --target=aarch64 -o test.aot test.wasm
Create AoT compiler with:
  target:        aarch64v8
  target cpu:    
  target triple: aarch64v8-pc-linux-gnu
  cpu features:  
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file test.aot was generated.
```

penneryu, 2024-02-19:
```
#if defined(_WIN32) || defined(_WIN32_) || defined(__APPLE__) \
    || defined(__MACH__)
        if (!option.target_abi) {
            LOG_VERBOSE("Set size level to 1 for Windows or MacOS AOT file");
            option.size_level = 1;
        }
#endif
    }
```
I found the problem, we should differentiate apple m1 chip

TianlongLiang, 2024-02-19:
Good idea, maybe we should differentiate intel-based and arm-based MacOS. For now, you can try adding a command line argument to set manually set the size-level now to see whether it works:

```shell
./wamrc --size-level=3 --target=aarch64 -o test.aot test.wasm
```

eloparco, 2024-02-19:
For AOT on M1, usually I compile with `wamrc --size-level=0 -o wasm.aot wasm.wasm` (as suggested by the LLVM error `LLVM ERROR: Only small, tiny and large code models are allowed on AArch64`):
```bash
Create AoT compiler with:
  target:        arm64
  target cpu:    apple-m1
  target triple: arm64-pc-linux-gnu
  cpu features:  
  opt level:     3
  size level:    0
  output format: AoT file
Compile success, file wasm.aot was generated.
```
The same is done in CI, where an M1 node (`macos-14`) is used: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/.github/workflows/compilation_on_macos.yml#L360