wasm-micro-runtime/wasm-micro-runtime #3165: Unexpected Behavior in JIT Mode

erxiaozhou, 2024-02-18:
# Build commands
I compile the code with commit id 8b8c59589dc725b04754018173463c388e7a29ea.
Platform: Ubuntu 20.04
CPU: amd64 
compile:

```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWASM_ENABLE_AOT=1 -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=0   -DCMAKE_INSTALL_PREFIX="xxx" -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1 -Bbuild
```

execute:
```shell
iwasm --heap-size=0 --llvm-jit -f to_test <tc_name>
```
# Cases

[iwasm_jit_select_exception.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/14323284/iwasm_jit_select_exception.zip)

```
(module
  (type (;0;) (func (result funcref)))
  (func (;0;) (type 0) (result funcref)
    ref.func 0
    ref.null func
    memory.size
    br_if 0 (;@0;)
    i32.const 1
    select (result funcref))
  (memory (;0;) 1 5)
  (export "_start" (func 0))
  (export "to_test" (func 0)))
```

# Actual behavior:
No exception
```
func:ref.null
```

# Expected behavior:
Exception
```
WASM module load failed: Error: invalid stack values with different type
```

TianlongLiang, 2024-02-19:
Hi, I think this bug is because of spec update, I used the low version wabt and there is the same exception. I will look into it.

erxiaozhou, 2024-02-19:
> Hi, I think this bug is because of spec update, I used the low version wabt and there is the same exception. I will look into it.
Wamr on different modes (e.g., interpreter mode, fast JIT mode) will not raise this exception.

wenyongh, 2024-02-21:
@erxiaozhou the issue was fixed with https://github.com/bytecodealliance/wasm-micro-runtime/pull/3169, please try again. Thanks for spotting the issue!

wenyongh, 2024-03-01:
Close the issue since it had been fixed.

closed 2024-03-01 by wenyongh