wasm-micro-runtime/wasm-micro-runtime #3168: unable to load module with undeclared function reference

peter-jerry-ye, 2024-02-20:
The following code may fail : 
```wasm
(module
 (type (func))
 (type (sub (struct (field (ref 0)))))
 (table funcref (elem $closure))
 (func $create (result (ref 1)) (struct.new 1 (ref.func $closure)))
 (func $closure (type 0)))
```

- if converted to wasm with wasmtime (`wasm-tools parse main.wat -o main.wasm`) or binaryen (`wasm-as main.wat -all -o main.wasm`) : `iwasm main.wasm` will fail with `WASM module load failed: undeclared function reference`
- if converted to wasm with [reference implementation](https://github.com/WebAssembly/gc/tree/main/interpreter) : `iwasm` will load the module without issue

It should be noticed though that the wasm generated by the wasmtime or binaryen still passes the validation of the reference implemenetation.

`iwasm` is built with `cmake -DWAMR_BUILD_GC=1 ..` on Mac with the latest commit b9db23b9

peter-jerry-ye, 2024-02-20:
The only difference between the generated codes (apart from the custom sections) is the section 9 (element section)

wasm-tools compiles to: `01 00 41 00 0b 01 01`

reference implementation compiles to: `01 06 00 41 00 0b 70 01 d2 01 0b`

wenyongh, 2024-02-26:
@peter-jerry-ye thanks for reporting the issue, this is due to that when GC is enabled, the type of table segment can be `(ref func)` now besides with `funcref`, while the original code only checks the latter. I submitted PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/3181 to fix it, please try again.

peter-jerry-ye, 2024-02-26:
@wenyongh It solves the issue. Thanks.

wenyongh, 2024-03-01:
The root cause is when GC is enabled, the type of table segment can be `(ref func)` now besides with `funcref`, while the original code only checks the latter.