wasm-micro-runtime/wasm-micro-runtime #3178: zephyr: old versions support

mkolchurin, 2024-02-22:
Do we need support for ancient versions of Zephyr that were released more than 5-6 years ago (`/* version 1.14.1 */`)? It might be much better to focus on newer versions (at least 2.7.0 LTS release, best 3.3.0)

[End-of-life releases](https://docs.zephyrproject.org/latest/releases/eol_releases.html)

The problem is that I'm trying to add new functionality (for example, support for `Alloc_With_System_Allocator` instead of `Alloc_With_Pool`), and it's hard to keep track of what has changed after so many releases.

wenyongh, 2024-02-26:
@mkolchurin it is really hard to support these ancient zephyr versions, I believe that they are rarely used by zephyr developers now, maybe we just don't add the version control in your Alloc_With_System_Allocator PR, and fix it when others report compilation error.

srberard, 2024-03-21:
I would say that it's not worth supporting directly.  According to the Zephyr project 1.14.x is no longer supported.  Here's the supported release table from Zephyr:
[image]

I'd rather see effort going to properly support the current releases of Zephyr.  There's a number of issues that are not working on Zephyr 3.5/3.6.