wasm-micro-runtime/wasm-micro-runtime #3184: build aot with debug on windows failed

xianjimli, 2024-02-28:
wasm-micro-runtime/core/iwasm/aot/debug/jit_debug.c

1. not found unistd.h
2. __attribute__(noinline) does't not work for MSVC 

core/iwasm/aot/debug/elf_parser.c

1. not found unistd.h

core/iwasm/common/wasm_runtime_common.c 

1. ZydisDecoderDecodeFull is changed in latest zydis
3. ZydisFormatterFormatInstruction is changed in latest zydis

OS: windows 10 
ARCH: arm64/amd64
Compiler: VC 2022

wenyongh, 2024-04-26:
Hi, thanks for spotting the issue, the aot debugger wasn't tested on windows yet, could you try the below patch:
```diff
diff --git a/core/iwasm/aot/debug/elf_parser.c b/core/iwasm/aot/debug/elf_parser.c
index 657f9530..7b0c57b8 100644
--- a/core/iwasm/aot/debug/elf_parser.c
+++ b/core/iwasm/aot/debug/elf_parser.c
@@ -7,7 +7,6 @@
 #include <assert.h>
 #include <fcntl.h>
 #include <stdlib.h>
-#include <unistd.h>
 #include <string.h>
 #include <errno.h>
 #include <stdbool.h>
diff --git a/core/iwasm/aot/debug/jit_debug.c b/core/iwasm/aot/debug/jit_debug.c
index 4b0e46f1..724a2c7e 100644
--- a/core/iwasm/aot/debug/jit_debug.c
+++ b/core/iwasm/aot/debug/jit_debug.c
@@ -24,7 +24,6 @@
 #include <stdio.h>
 #include <assert.h>
 #include <stdlib.h>
-#include <unistd.h>
 #include <string.h>
 #include <errno.h>
 #include <stdbool.h>
@@ -56,6 +55,12 @@ typedef struct JITDescriptor {
     JITCodeEntry *first_entry_;
 } JITDescriptor;

+#if defined(_WIN32) || defined(_WIN32_)
+#define attribute_noinline __declspec(noinline)
+#else
+#define attribute_noinline __attribute__((noinline))
+#endif
+
 /* LLVM has already define this */
 #if (WASM_ENABLE_WAMR_COMPILER == 0) && (WASM_ENABLE_JIT == 0)
 /**
@@ -63,9 +68,9 @@ typedef struct JITDescriptor {
  * To prevent GCC from inlining or removing it we place noinline attribute
  * and inline assembler statement inside.
  */
-void __attribute__((noinline)) __jit_debug_register_code();
+void attribute_noinline __jit_debug_register_code();

-void __attribute__((noinline)) __jit_debug_register_code()
+void attribute_noinline __jit_debug_register_code()
 {
     int x;
     *(char *)&x = '\0';
```

xianjimli, 2024-04-26:
OK, Thank you

wenyongh, 2024-04-29:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3370