wasm-micro-runtime/wasm-micro-runtime #3185: dump translated bytecode for fast interpreter

yamt, 2024-02-28:
for debugging etc, it would be great to have a way to dump translated bytecode in a human readable format.
i feel `WASM_DEBUG_PREPROCESSOR` is a bit too primitive.

wenyongh, 2024-03-01:
Hi, it may be a little complex to dump the pre-compiled bytecode with well-formed human readable format, is it good to dump the opcode mnemonics instead of the hex value? Here is the patch:
```diff
diff --git a/core/iwasm/interpreter/wasm_loader.c b/core/iwasm/interpreter/wasm_loader.c
index 8bf2ed92..2a713f3a 100644
--- a/core/iwasm/interpreter/wasm_loader.c
+++ b/core/iwasm/interpreter/wasm_loader.c
@@ -7604,6 +7604,9 @@ fail:
 #if WASM_ENABLE_FAST_INTERP != 0

 #if WASM_DEBUG_PREPROCESSOR != 0
+#define HANDLE_OPCODE(opcode) #opcode
+DEFINE_GOTO_TABLE(const char *, opcode_names);
+#undef HANDLE_OPCODE
 #define LOG_OP(...) os_printf(__VA_ARGS__)
 #else
 #define LOG_OP(...) (void)0
@@ -8514,7 +8517,7 @@ fail:
 #define emit_label(opcode)                                      \
     do {                                                        \
         wasm_loader_emit_ptr(loader_ctx, handle_table[opcode]); \
-        LOG_OP("\nemit_op [%02x]\t", opcode);                   \
+        LOG_OP("\nemit_op [%s]\t", opcode_names[opcode]);       \
     } while (0)
 #define skip_label()                                            \
     do {                                                        \
```
And the output is like:
```wast
Processing func | [0] params | [0] locals | [0] return

emit_op [WASM_OP_END]
delete last op

emit_op [WASM_OP_RETURN]
emit_op [WASM_OP_END]
delete last op

emit_op [WASM_OP_RETURN]
Processing func | [0] params | [4] locals | [0] return

emit_op [WASM_OP_GET_GLOBAL]    0   4
emit_op [WASM_OP_I32_CONST]
delete last op
#### new const [1]: 16

emit_op [WASM_OP_I32_SUB]   -1  4   5
emit_op [WASM_OP_TEE_LOCAL]
delete last op

emit_op [EXT_OP_TEE_LOCAL_FAST] 0   5
emit_op [WASM_OP_SET_GLOBAL]    0   5
emit_op [WASM_OP_CALL]  47
emit_op [WASM_OP_I32_CONST]
delete last op
#### new const [2]: 3

emit_op [WASM_OP_SET_LOCAL]
delete last op
```

yamt, 2024-03-04:
> is it good to dump the opcode mnemonics instead of the hex value?

yes, it's nice.
i'm also thinking to process "delete last op" automatically by having a one-line buffer.