wasm-micro-runtime/wasm-micro-runtime #3187: C++ related issues in dwarf_extractor.cpp

yamt, 2024-02-29:
when compiling a C++ module, in lldb_function_to_function_dbi,

* `function_args` doesn't seem to include the "this" pointer. on the other hand, `variable_list` contains it. it causes the "function args number dismatch" message and then ParamType overrun, which likely involves a crash.
* `variable.GetName()` seems to return NULL for unnamed arg. it causes a NULL dereference.

yamt, 2024-03-07:
btw, i wonder why dwarf_extractor.cpp uses the LLDB API, rather than, say, llvm/DebugInfo/DWARF/ api.
@wenyongh do you remember?

wenyongh, 2024-03-07:
> btw, i wonder why dwarf_extractor.cpp uses the LLDB API, rather than, say, llvm/DebugInfo/DWARF/ api. @wenyongh do you remember?

The source debugger feature was implemented by Ant group long time ago, I just cannot remember the details now. Maybe the LLDB API provides more capability?

yamt, 2024-03-08:
a quick research on how other projects are reading the dwarf section in wasm:

binaryen: llvm DWARFContext https://github.com/WebAssembly/binaryen/blob/d6c5e4ab15df271521df7b35665c7463b2c490ca/src/wasm/wasm-debug.cpp#L96-L116

cranelift: https://github.com/gimli-rs/gimli (something written in rust)

TianlongLiang, 2024-03-21:
It seems that AOT debugging now only has limited functionality. It was aimed to support C first It hardly covers any additional C++ features. Thus the current function-to-function debug symbol mapping only supports C styles function. And I think that's why it error when it comes to Cpp compiled wasm or Rust compiled wasm

yamt, 2024-04-01:
> It seems that AOT debugging now only has limited functionality. It was aimed to support C first It hardly covers any additional C++ features. Thus the current function-to-function debug symbol [...]

i agree.

i tend to think a fix involves a major surgery like switching from lldb api to llvm api to read dwarf. how do you think?

TianlongLiang, 2024-04-03:
I think it's feasible, but as you said it would be non-trivial changes. BTW, besides the extractor, will there also be some major modifications for code emission and file generation?

yamt, 2024-04-03:
> I think it's feasible, but as you said it would be non-trivial changes. BTW, besides the extractor, will there also be some major modifications for code emission and file generation?

i guess it will be mostly contained in dwarf_extractor.cpp.

TianlongLiang, 2024-04-03:
> i guess it will be mostly contained in dwarf_extractor.cpp.

Got it, thanks