wasm-micro-runtime/wasm-micro-runtime #3193: addr2line.py doesn't show correct location information

eloparco, 2024-03-01:
I was trying to use the [addr2line.py](https://github.com/bytecodealliance/wasm-micro-runtime/blob/56352441691243facca9f40ce362b8f473315213/test-tools/addr2line/addr2line.py) tool, but it doesn't seem to report the right location info.

I have a simple c file
```c
int
c(int n)
{
    __builtin_trap();
}

int
b(int n)
{
    n += 3;
    return c(n);
}

int
a(int n)
{
    return b(n);
}

int
main()
{
    int i = 5;
    a(i);

    return 0;
}
```
that I compile with `/opt/wasi-sdk/bin/clang -g -target wasm32-wasi trap.c -o trap.wasm`.
I get the stack trace from the latest `main` branch:
```text
#00: 0x004a - $f3
#01: 0x00a9 - $f4
#02: 0x0070 - $f5
#03: 0x0096 - $f6
#04: 0x0088 - _start
```
and feed it to the script, but I don't get the location info. The addresses reported by `iwasm` don't match the ones shown by `llvm-dwarfdump`. 
Am I doing something wrong?

lum1n0us, 2024-03-01:
Nice catch. It is a bug of addr2line.py.

lum1n0us, 2024-03-01:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3194

eloparco, 2024-03-01:
Hi @lum1n0us, thanks for the quick reply! I tried the fix in #3194, now I get the backtrace but it's not correct:
```bash
#00: 0x004a - $f3
#01: 0x00a9 - $f4 (FILE:trap.c LINE:   10 COLUMN:  7 FUNC:b)
#02: 0x0070 - $f5 (FILE:trap.c LINE:    3 COLUMN:  0 FUNC:c)
#03: 0x0091 - $f6 (FILE:trap.c LINE:    9 COLUMN:  0 FUNC:b)
#04: 0x0088 - _start (FILE:trap.c LINE:    9 COLUMN:  0 FUNC:b)
```
I was comparing the script to https://github.com/bytecodealliance/wasm-tools/blob/main/src/bin/wasm-tools/addr2line.rs, but couldn't find what the problem is.

lum1n0us, 2024-03-04:
#3199 

```
Before running this script, please make sure:
  - the wasm file is compiled with debug info. (like: clang -g) 
  - the call-stack dump is generated by iwasm
  - iwasm is compiled with -DWAMR_BUILD_DUMP_CALL_STACK=1
  - iwasm isn't running under fast-interp mode. -DWAMR_FAST_INTERP=0
  - if using .aot, the aot file is generated with `--enable-dump-call-stack`
```

eloparco, 2024-03-05:
Thanks for the fix, closing the issue.
I have some changes that I want to add to the script, I'll raise a PR once #3199 is merged

closed 2024-03-05 by eloparco