wasm-micro-runtime/wasm-micro-runtime #3208: Issue when running examples for ESP 32 with WAMR

TAOFOR4, 2024-03-07:
Hi,

I am trying to run the samples from the repo on esp c6. But some issues occur.

[image]

TAOFOR4, 2024-03-07:
A little update. After I set some compiler options in cmake file it shows:
[image]

wenyongh, 2024-03-08:
Hi, thanks for reporting the issue, could you try the below patch:
```diff
diff --git a/core/shared/platform/esp-idf/espidf_memmap.c b/core/shared/platform/esp-idf/espidf_memmap.c
index 9f3ec47a..6b1b6f04 100644
--- a/core/shared/platform/esp-idf/espidf_memmap.c
+++ b/core/shared/platform/esp-idf/espidf_memmap.c
@@ -59,6 +59,12 @@ os_mmap(void *hint, size_t size, int prot, int flags, os_file_handle file)
     }
 }

+void *
+os_mremap(void *old_addr, size_t old_size, size_t new_size)
+{
+    return os_mremap_slow(old_addr, old_size, new_size);
+}
+
 void
 os_munmap(void *addr, size_t size)
 {
diff --git a/core/shared/platform/esp-idf/platform_internal.h b/core/shared/platform/esp-idf/platform_internal.h
index 70c4fe7b..91a3f801 100644
--- a/core/shared/platform/esp-idf/platform_internal.h
+++ b/core/shared/platform/esp-idf/platform_internal.h
@@ -109,6 +109,8 @@ typedef unsigned int korp_sem;
 #define DT_LNK DTYPE_LINK
 #define DT_SOCK DTYPE_SOCK

+#define os_getpagesize getpagesize
+
 typedef int os_file_handle;
 typedef DIR *os_dir_stream;
 typedef int os_raw_file_handle;
```

TAOFOR4, 2024-03-08:
Hi, thx for the quick reply! I tried your patch. 
## For the example from WAMR repo it still complains. 
[image]

wenyongh, 2024-03-09:
Could you try:
```diff
diff --git a/core/shared/platform/esp-idf/espidf_platform.c b/core/shared/platform/esp-idf/espidf_platform.c
index 8d3a9b87..9c0d02e6 100644
--- a/core/shared/platform/esp-idf/espidf_platform.c
+++ b/core/shared/platform/esp-idf/espidf_platform.c
@@ -234,7 +234,7 @@ unlinkat(int fd, const char *path, int flag)
 }

 int
-utimensat(int fd, const char *path, const struct timespec *ts, int flag)
+utimensat(int fd, const char *path, const struct timespec ts[2], int flag)
 {
     errno = ENOSYS;
     return -1;
@@ -257,7 +257,7 @@ ftruncate(int fd, off_t length)
 #endif

 int
-futimens(int fd, const struct timespec *times)
+futimens(int fd, const struct timespec times[2])
 {
     errno = ENOSYS;
     return -1;
```

TAOFOR4, 2024-03-09:
> Could you try:
>
> ```diff
> diff --git a/core/shared/platform/esp-idf/espidf_platform.c b/core/shared/platform/esp-idf/espidf_platform.c
> index 8d3a9b87..9c0d02e6 100644
> --- [...]

Hi Thank you! It works but new issue occurs as below:
[image]

wenyongh, 2024-03-11:
Hi, `depth` should have been set in L11433 `read_leb_uint32(p, p_end, depth)`, not sure why the compiler reports the warning. Please initialize it with 0:
```diff
diff --git a/core/iwasm/interpreter/wasm_loader.c b/core/iwasm/interpreter/wasm_loader.c
index 8bf2ed92..d9ea015b 100644
--- a/core/iwasm/interpreter/wasm_loader.c
+++ b/core/iwasm/interpreter/wasm_loader.c
@@ -11412,7 +11412,7 @@ re_scan:

             case WASM_OP_BR_TABLE:
             {
-                uint32 depth, default_arity, arity = 0;
+                uint32 depth = 0, default_arity, arity = 0;
                 BranchBlock *target_block;
                 BlockType *target_block_type;
 #if WASM_ENABLE_FAST_INTERP == 0
```

TAOFOR4, 2024-03-11:
Hi, yeah its weird. And now it shows:
[image]

TAOFOR4, 2024-03-14:
> @TAOFOR4 Could you please try #3224?

Hi, it works well now :) Thx for help! Btw, will WAMR add support for ESP C6? I noticed that right now it only supports ESP32 and ESP32 C3. Thank you!

wenyongh, 2024-03-15:
> > @TAOFOR4 Could you please try #3224?
> 
> Hi, it works well now :) Thx for help! Btw, will WAMR add support for ESP C6? I noticed that right now it only supports ESP32 and ESP32 C3. Thank you!

Welcome, currently no plan to support ESP C6 yet. Maybe you can have a try and help commit the code? It would be greatly appreciated.

TAOFOR4, 2024-03-15:
> Welcome, currently no plan to support ESP C6 yet. Maybe you can have a try and help commit the code? It would be greatly appreciated.

My pleasure. I will look into that part.

closed 2024-03-15 by TAOFOR4