wasm-micro-runtime/wasm-micro-runtime #3212: Stack frame address issue in AOT mode

eloparco, 2024-03-08:
When printing the stack trace in AOT, the first stack frame has address `0x000`, e.g.
```text
#00: 0x0000 - $f5
#01: 0x01ac - $f6
#02: 0x01fa - $f7
#03: 0x0265 - $f8
#04: 0x2365 - $f15
#05: 0x0119 - _start
```
How to reproduce: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/debug-tools/symbolicate.sh#L18-L23.

It doesn't always happen (with other wasm/aot files), I tried compiling some Rust code and I wasn't getting the problem.

wenyongh, 2024-03-09:
@eloparco please try #3213.

eloparco, 2024-03-11:
Thanks for the quick fix, it works now!
```bash
#00: 0x0158 - $f5
#01: 0x01ac - $f6
#02: 0x01fa - $f7
#03: 0x0265 - $f8
#04: 0x2365 - $f15
#05: 0x0119 - _start
```

closed 2024-03-11 by eloparco