wasm-micro-runtime/wasm-micro-runtime #3223: Report a possible bug related to address validation

AuYang261, 2024-03-13:
I think there might be a bug in `tensor_data_app_native` function in `core/iwasm/libraries/wasi-nn/src/utils/wasi_nn_app_native.c` file: the last parameter passed to `wasm_runtime_validate_app_addr` should be `total_elements` * `size of each element`, which depending on `input_tensor_wasm->type`.

Because the parameter `total_elements` is the number of the elements in array pointed by `input_tensor_wasm->data_offset`, while the length of the array in bytes should be the number of elements * size of each element.

I wonder if my understanding is wrong. Thanks.

wenyongh, 2024-03-14:
@tonibofarull I am not sure whether it is a bug, could you help look into it? Thanks a lot.

yamt, 2025-08-28:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/4534 fixed this.

closed 2025-08-28 by yamt