wasm-micro-runtime/wasm-micro-runtime #3229: Does WAMR support wasi_ephemeral_nn module?

AuYang261, 2024-03-14:
I try to run wasm file with wasi-nn imported built from rust, but WAMR print that cannot import functions in `wasi_ephemeral_nn` module and failed when try to call `load`:

```
warning: failed to link import function (wasi_ephemeral_nn, load)
warning: failed to link import function (wasi_ephemeral_nn, init_execution_context)
warning: failed to link import function (wasi_ephemeral_nn, compute)
warning: failed to link import function (wasi_ephemeral_nn, get_output)
warning: failed to link import function (wasi_ephemeral_nn, set_input)
...
Exception: failed to call unlinked import function (wasi_ephemeral_nn, load)
```

It seems that the wasm file built from rust requires `wasi_ephemeral_nn` module instead of `wasi_nn`. Does WAMR support `wasi_ephemeral_nn` module? If not currently, can I add the `wasi_ephemeral_nn` module, just as `wasi-nn` was written, and following the wasi-ephemeral-nn interface defined in rust?

AuYang261, 2024-03-15:
I have added `wasi_ephemeral_nn` module imitating `wasi_nn` in [my fork](https://github.com/AuYang261/wasm-micro-runtime/commit/fc4e6aba9241937584cc6a64cc662b95ba599495#diff-bc01a60a3118c887ab3f0742e4059f0b8fdecd17af557f591f56e1d4d4a3f39c) to meet the needs of my application.

Should I create a PR to merge the part of `wasi_ephemeral_nn` into main branch? Or is there no need to support `wasi_ephemeral_nn` in WAMR for the reason `wasi_ephemeral_nn` is temporary?

AuYang261, 2024-03-17:
I noticed it's similar to what #3103 mentioned.

wenyongh, 2024-03-18:
> I have added `wasi_ephemeral_nn` module imitating `wasi_nn` in [my [...]

@AuYang261 thanks a lot, that would be great! Could you add cmake variable `WAMR_BUILD_WASI_EPHEMERAL_NN` and compiler macro `WASM_ENABLE_WASI_EPHERMERAL` to control the build script and source code?
BTW, `wasi_ephemeral_nn` is enabled only when `wasi_nn` is enabled, right?

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/build-scripts/config_common.cmake#L422-L432

AuYang261, 2024-03-18:
> @AuYang261 thanks a lot, that would be great! Could you add cmake variable `WAMR_BUILD_WASI_EPHEMERAL_NN` and compiler macro `WASM_ENABLE_WASI_EPHERMERAL` to control the build script and source [...]

You are right. I will add cmake variable and compiler macro to control whether to enable `wasi_ephemeral_nn` when `wasi_nn` is enabled and would like to create a PR.

closed 2024-03-23 by AuYang261