wasm-micro-runtime/wasm-micro-runtime #3238: Allow defining the ANDROID_ABI and WAMR_BUILD_TARGET for Android building

daniel7grant, 2024-03-18:
For our use-case, we need to build WAMR for multiple Android platforms, which means that we have to change the ANDROID_ABI and WAMR_BUILD_TARGET variables. Currently these can only be set by modifying `CMakeLists.txt` in `product-mini/platforms/android`. This is harder to do when it comes to automatic building (CI), parallelization, etc.

I'd make two modification to be able to set these parameters with `-D` defines:

```cmake
if (NOT DEFINED ANDROID_ABI)
    set (ANDROID_ABI "x86")
endif ()
```

and:

```cmake
if (NOT DEFINED WAMR_BUILD_TARGET)
    set (WAMR_BUILD_TARGET    "X86_32")
endif ()
```

With these changes, the cross compile could be called without ever changing the files:

```bash
cmake -DANDROID_ABI=armeabi-v7a -DWAMR_BUILD_TARGET=ARMV7A
```

What do you think? I could create a Pull Request for this and the documentation update.

wenyongh, 2024-03-19:
Hi, yes, it is not a good way to hardcode the configurations in the cmake file, I remember it was written that way due to the GitHub CI which only supports android 32-bit. I submitted PR #3239 to rewrite the cmake file and CI file, could you help have a view? Thanks.

daniel7grant, 2024-03-19:
Thanks for your work, I tried and it seems good to me.

Simultaneously I started working on this problem (https://github.com/bytecodealliance/wasm-micro-runtime/compare/main...daniel7grant:wasm-micro-runtime:feature/android-build-options), but your `CMakeLists.txt` seems to be more useful. The only thing I would add from my branch is the documentation: to add a note in the `platform-mini/README.md` that the flags should be used instead of modifying the `CMakeLists.txt` file.

wenyongh, 2024-03-19:
OK, I updated the document according to you branch and made small changes. Please check whether more need to be added or changed?

daniel7grant, 2024-03-19:
I don't see anything else. Thank you for your help.

wenyongh, 2024-03-20:
Yes, I did some test, it is not easy to set `WAMR_BUILD_TARGET` according to the `CMAKE_SYSTEM_PROCESSOR`: `CMAKE_SYSTEM_PROCESSOR` is available only when `project (iwasm)` is added in the CMakeLists.txt, but if the latter is added before setting `ANDROID_ABI` and `CMAKE_TOOLCHAIN_FILE`, the c/c++ compiler will be set to GNU compiler instead of NDK's compiler, and compilation errors will be reported. So I updated the PR to enforce the developer to set `WAMR_BUILD_TARGET` explicitly, and use it to set `ANDRDOI_ABI` if it isn't passed from the command line, and then pass it `android.toolchain.cmake`, which checks `ANDRDOI_ABI` and sets `CMAKE_ANDROID_ARCH_ABI` accordingly:
https://android.googlesource.com/platform/ndk/+/master/build/cmake/android.toolchain.cmake#110

Please help review and try again, let's merge the PR if it works fine for you.

daniel7grant, 2024-03-20:
I tried it and it seems to be working and producing valid libs on every platform. Thank you, feel free to merge it.

closed 2024-03-21 by wenyongh