wasm-micro-runtime/wasm-micro-runtime #3267: Troubleshooting of Logging Interpreter on ESP32/QEMU with Zephyr RTOS

OkannShn, 2024-03-31:
I have an ESP32 board and I am interested in running a WebAssembly (WASM) application on the Zephyr RTOS. To understand the process of loading and executing the WASM binary, I would like some guidance on monitoring the workflow. Instead of using a debugger, which can be complex to follow step-by-step, I prefer to track the process through log messages that detail how the interpreter operates. Somehow `printf `does not work.

### Example
[image]

While ı am debugging, ı observe that program enters `wasm_runtime_invoke_native`,  but after ı compile and flash the program with `printf `statement. ı can not see message properly. ı only see a few characters even if program enters here more than once.

### Board
Link to board https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html
```
 cd wasm-micro-runtime/product-mini/platforms/zephyr/simple
 west build -b esp32_devkitc_wroom . -p always ---DWAMR_BUILD_TARGET=XTENSA
 west flash
```

I tried whether ı can observe log messages with `qemu_cortex_a53`. It does not work again.

This might be a very simple question, I'm a newbie to wasm. It would be great to see any help. Thank you

TianlongLiang, 2024-04-03:
Hi, could you try replacing `printf` with `os_printf`? If that doesn't work, I suspect that it may be due to your board's buffering or console configuration. You can try a Hello World C program to test whether it can print successfully to verify whether it's a configuration issue.

OkannShn, 2024-04-03:
```
west build -b qemu_cortex_a53 . -p always -- -DWAMR_BUILD_TARGET=AARCH64
west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-a53
iwasm_main
45--- 54 messages dropped ---
8
buf: 1234
elapsed: 0
```

I inserted a few `os_printf`, As far as i understand it does not work after I initialized the working env with `wasm_runtime_full_init`. However, I observe that not printed messages are seen as dropped messages. The question then becomes how to prevent message drops?

TianlongLiang, 2024-04-03:
Maybe try building WAMR with Debug mode using CMake flags `cmake -DCMAKE_BUILD_TYPE=Debug`, so the west build command should be:
`west build -b esp32_devkitc_wroom . -p always -- -DWAMR_BUILD_TARGET=XTENSA -DCMAKE_BUILD_TYPE=Debug` 
I think maybe the non-debug mode disables some macro so that it won't print inside the vmcore https://github.com/bytecodealliance/wasm-micro-runtime/blob/4806e4e298c42bee9d35c542d8a7b1ceaef86c1c/core/shared/platform/zephyr/zephyr_platform.c#L135

OkannShn, 2024-04-03:
Unfortunately, it did not work for `qemu_cortex_a53`, I will also try for `esp32_devkitc_wroom `, but ı expect similar output

OkannShn, 2024-04-14:
Hi,

I found the cause of the issue. It looks like the Zephyr logging mechanism causes it. You can check https://github.com/zephyrproject-rtos/zephyr/issues/14903.
I have added `CONFIG_LOG_BUFFER_SIZE=4096` into  `wasm-micro-runtime/product-mini/platforms/zephyr/simple`, and it works. Indeed, I did not expect to solve the issue by doing this because the log mechanism is different than basic `prinft`. Does it mean that Wamr redirects all the prints to Zephyr log when Zephyr is selected as a platform?

wenyongh, 2024-04-29:
@OkannShn thanks for the reminding, I added `CONFIG_LOG_BUFFER_SIZE=4096` in https://github.com/bytecodealliance/wasm-micro-runtime/pull/3370.