wasm-micro-runtime/wasm-micro-runtime #3277: What is the difference between clang and em++?

kamylee, 2024-04-03:
I compiled the libjsoncpp.a library using Emscripten. When I tried to compile wasm using clang and linked this library, it resulted in errors at runtime.
**cmd:**
`clang -O3 -nostdlib \
-pthread \
-Wl,--shared-memory,--max-memory=131072 \
-o test.wasm test.cpp ./lib/libjsoncpp.a \
-Wl,--export=init -Wl,--export=onMessage \
-Wl,--export=__heap_base -Wl,--export=__data_end \
-Wl,--no-entry -Wl,--strip-all -Wl,--allow-undefined`
**err:**
warning: failed to link import function (env, _Znwm)
warning: failed to link import function (env, _ZdlPv)
warning: failed to link import function (env, _ZNSt12length_errorD1Ev)
warning: failed to link import function (env, _ZNSt11logic_errorC2EPKc)
warning: failed to link import function (env, _ZNSt3__212basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE9__grow_byEmmmmmm)

**However,** when I compiled the wasm using em++, it could parse JSON correctly at runtime.
**cmd:**
`em++ -g -O3 test.cpp lib/libjsoncpp.a -o test.wasm --no-entry -s WASM=1 -s ERROR_ON_UNDEFINED_SYMBOLS=0  -s EXPORTED_FUNCTIONS=_init,_onMessage`
**result:**
[image]

Is it because clang uses wasi-sdk, while em++ uses Emscripten? Does wasi-sdk not support jsoncpp?
How to solve? Thank you!

kamylee, 2024-04-03:
Thank you for your reply! I tried using clang++,But it ditn't work still. still the same error.

TianlongLiang, 2024-04-03:
What if you try to remove `-nostdlib`? If that doesn't work, then I think maybe you are right at first, it could be wasi-sdk problem

Rochet2, 2024-04-03:
The errors you get seem to be from exception related imports that cannot be resolvedas the runtime does not provide the required imports for the module.

If you look at the readme on [wasi-sdk main page](https://github.com/WebAssembly/wasi-sdk) you can see that it does not support exceptions. Also note that since the exceptions come from std, using nostdlib will possibly make the missing classes into imports that then fail as you try to run the program.
> **This repository does not yet support C++ exceptions. C++ code is supported only with -fno-exceptions for now.**

Emscripten by default does not support exceptions either. However, it will allow the code to compile and run. It will abort the entire program when an exception occurs by default. The docs say that there is some kind of support possible, but I never got it to work (tried for few hours to get lua to compile to wasm with exceptions).
See [emscripten docs on exceptions](https://emscripten.org/docs/porting/exceptions.html).

I am not aware of non-emscripten exception tolerant/able compilers for wasm, but then again I have not looked too much.