wasm-micro-runtime/wasm-micro-runtime #3279: wasm_interp_call_func_native currupts frame->ip, causes SIGSEGV with native tail-calls

SBKarr, 2024-04-03:
### Subject of the issue

`wasm_interp_call_func_native` [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/b4cab84e5f6006185f44db2c66fe389850e5f4b9/core/iwasm/interpreter/wasm_interp_fast.c#L5854) uses `prev_frame` as frame origin. Next allocated frame will share same memory space as `frame`  in interpreter.  Then, it's ip will be set to NULL [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/b4cab84e5f6006185f44db2c66fe389850e5f4b9/core/iwasm/interpreter/wasm_interp_fast.c#L1176). As a result, [UPDATE_ALL_FROM_FRAME](https://github.com/bytecodealliance/wasm-micro-runtime/blob/b4cab84e5f6006185f44db2c66fe389850e5f4b9/core/iwasm/interpreter/wasm_interp_fast.c#L5860C13-L5860C34) restores `frame_ip = NULL`, that causes crash on SIGSEGV on continue.

### Test case

In call of _initialize. Specifically, in `stappler_wasm_wasm_debug_print` line 43.
[module4.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/14853332/module4.txt)

### Your environment

* Ubuntu 22.04, x86_64, WASI SDK 21
* WAMR 1.3.2 + WAMR  from /release/1.3.x

### Steps to reproduce

- Build WAMR with tail-call support
- Define native functon like this:
```c
static void StapplerWasmDebugPrint(wasm_exec_env_t exec_env, uint32_t ptr, uint32_t size) {
	auto mod = wasm_runtime_get_module_inst(exec_env);
	auto sptr = (const char *)wasm_runtime_addr_app_to_native(mod, ptr);

	printf(sptr);
	printf("\n");
}

static NativeSymbol stapper_wasm_symbols[] = {
	NativeSymbol{"debug-print", (void *)&StapplerWasmDebugPrint, "(ii)", NULL},
};

static void registerWasmModule() {
	wasm_runtime_register_natives("stappler:wasm/wasm", stapper_wasm_symbols, 1);
}
```
- Run code from test case, call `_initialize`

### Expected behavior

Should print `TEST CTOR` from native code

### Actual behavior

`SIGSEGV`

### Extra Info

Fixed this by adding `SYNC_ALL_TO_FRAME();` after `wasm_interp_call_func_native`, but not sure, if it's correct solution.

wenyongh, 2024-04-06:
@SBKarr Thanks for spotting issue! I checked the case, it should be caused by calling native function by opcode return_call, which is introduced by wasm tail call feature. Since the tail call uses the current frame and it is freed first and then allocated and overwritten (frame->ip = NULL) by wasm_interp_call_func_native, the later recovering `frame_ip` from its ip field is invalid. But I think after calling native function, the current `frame_ip`, `frame_ref` (and `prev_frame`) wasn't changed and are effective, no need to commit them to frame and restore them from frame again. Could you help check whether the below patch is better?
```diff
diff --git a/core/iwasm/interpreter/wasm_interp_classic.c b/core/iwasm/interpreter/wasm_interp_classic.c
index ca972fd4..43e71913 100644
--- a/core/iwasm/interpreter/wasm_interp_classic.c
+++ b/core/iwasm/interpreter/wasm_interp_classic.c
@@ -1509,6 +1509,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
     WASMStringviewIterObjectRef stringview_iter_obj;
 #endif
 #endif
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+    bool is_return_call = false;
+#endif
 #if WASM_ENABLE_MEMORY64 != 0
     /* TODO: multi-memories for now assuming the memory idx type is consistent
      * across multi-memories */
@@ -6227,6 +6230,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
                 frame_ip = frame->ip;
                 frame_sp = frame->sp;
                 frame_csp = frame->csp;
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+                is_return_call = false;
+#endif
                 goto call_func_from_entry;
             }

@@ -6320,6 +6326,7 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
         }
         FREE_FRAME(exec_env, frame);
         wasm_exec_env_set_cur_frame(exec_env, prev_frame);
+        is_return_call = true;
         goto call_func_from_entry;
     }
 #endif
@@ -6333,6 +6340,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
         }
         SYNC_ALL_TO_FRAME();
         prev_frame = frame;
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+        is_return_call = false;
+#endif
     }

     call_func_from_entry:
@@ -6394,9 +6404,14 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
                                              prev_frame);
             }

-            prev_frame = frame->prev_frame;
-            cur_func = frame->function;
-            UPDATE_ALL_FROM_FRAME();
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+            if (!is_return_call)
+#endif
+            {
+                prev_frame = frame->prev_frame;
+                cur_func = frame->function;
+                UPDATE_ALL_FROM_FRAME();
+            }

             /* update memory size, no need to update memory ptr as
                it isn't changed in wasm_enlarge_memory */
diff --git a/core/iwasm/interpreter/wasm_interp_fast.c b/core/iwasm/interpreter/wasm_interp_fast.c
index 21412046..599f7056 100644
--- a/core/iwasm/interpreter/wasm_interp_fast.c
+++ b/core/iwasm/interpreter/wasm_interp_fast.c
@@ -1501,6 +1501,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
     WASMStringviewIterObjectRef stringview_iter_obj;
 #endif
 #endif
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+    bool is_return_call = false;
+#endif

 #if WASM_ENABLE_LABELS_AS_VALUES != 0
 #define HANDLE_OPCODE(op) &&HANDLE_##op
@@ -5618,6 +5621,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
             {
                 frame = prev_frame;
                 frame_ip = frame->ip;
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+                is_return_call = false;
+#endif
                 goto call_func_from_entry;
             }

@@ -5766,6 +5772,7 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
         FREE_FRAME(exec_env, frame);
         frame_ip += cur_func->param_count * sizeof(int16);
         wasm_exec_env_set_cur_frame(exec_env, (WASMRuntimeFrame *)prev_frame);
+        is_return_call = true;
         goto call_func_from_entry;
     }
 #endif /* WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0 */
@@ -5838,6 +5845,9 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
         }
         SYNC_ALL_TO_FRAME();
         prev_frame = frame;
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+        is_return_call = false;
+#endif
     }

     call_func_from_entry:
@@ -5855,9 +5865,14 @@ wasm_interp_call_func_bytecode(WASMModuleInstance *module,
                                              prev_frame);
             }

-            prev_frame = frame->prev_frame;
-            cur_func = frame->function;
-            UPDATE_ALL_FROM_FRAME();
+#if WASM_ENABLE_TAIL_CALL != 0 || WASM_ENABLE_GC != 0
+            if (!is_return_call)
+#endif
+            {
+                prev_frame = frame->prev_frame;
+                cur_func = frame->function;
+                UPDATE_ALL_FROM_FRAME();
+            }

             /* update memory size, no need to update memory ptr as
                it isn't changed in wasm_enlarge_memory */
```
```

wenyongh, 2024-04-06:
#3283

wenyongh, 2024-04-07:
Before tail calling, the current frame was freed and the exec_env's cur_frame was set as prev_frame:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/4ef724bbfffab1c7761b2674c483a8e7d7c840b8/core/iwasm/interpreter/wasm_interp_fast.c#L5766-L5768
then during tail calling to import function or native function, a new frame was allocated and then freed (see wasm_interp_call_func_native), so after tail calling, the `current frame` doesn't take effect again, and we should restore variables from `prev_frame` instead. If not, core dump may occur if we add an extra return_call in your case:
```wast
(func $stappler_wasm_wasm_debug_print (type 0) (param i32)
    local.get 0
    i32.load
    local.get 0
    i32.load offset=4
    return_call $__wasm_import_stappler_wasm_wasm_debug_print
    return_call $__wasm_import_stappler_wasm_wasm_debug_print
    )
```
I updated PR #3283, please help have a view and try, thanks.

SBKarr, 2024-04-07:
Ok, this solution works in my case, thanks

wenyongh, 2024-04-08:
Welcome, I merged the PR.

closed 2024-04-08 by wenyongh