wasm-micro-runtime/wasm-micro-runtime #3292: unexpected type mismatch with gc

peter-jerry-ye, 2024-04-09:
### Subject of the issue

`iwasm` reported

> WASM module load failed: type mismatch: expect (ref null ht) but got other

While the reference implementation consider the wasm valid

### Test case

```wasm
(module
(type $f (sub (struct)))
(type $f/2 (sub (struct)))
(func $f/1 (param (ref null $f)))
(func $f/2 (call $f/1 (ref.null $f/2) ))
)
```

### Your environment

* MacOS
* commit dacb3c4105adcd3d7db90516dc2ddf7118a8d394, M1 arm 64, iwasm with GC enabled

### Steps to reproduce

Tell us how to reproduce this bug or defect.

### Expected behavior

The module loads

### Actual behavior

> WASM module load failed: type mismatch: expect (ref null ht) but got other

### Extra Info

Error occurs if compiled with `wasm-tools` (version 1.202.0) but not with `binaryen` (wasm-as version 117)

peter-jerry-ye, 2024-04-12:
If I understand correctly, the function `wasm_type_is_supers_of` in `gc_type.c` does not seem to handle structural equivalence.

The reason why `binaryen` works is that `binaryen` will merge same types into one while `wasm-tools` will preserve them.

wenyongh, 2024-04-15:
@peter-jerry-ye thanks for reporting the issue, since the GC MVP proposal was evolving quickly and it is not easy to understand some of the check rules, there are some defects in the GC subtyping check. I submitted PR #3317 to enhance it, could you please try again?

peter-jerry-ye, 2024-04-15:
Hi, I found a counter example here. The following code will have 'function type mismatch' while it shouldn't.
```Wasm
(module
(rec
 (type $t1-sig (func))
 (type $t1 (sub (struct (field (mut (ref null $t1-sig)))))))
(rec
 (type $t2-sig (func))
 (type $t2 (sub (struct (field (mut (ref null $t2-sig)))))))

(func $f1 (call_ref $t2-sig (struct.get $t2 0 (ref.as_non_null (ref.null $t2)))))
)
```

It also happens when it's parsed with `wasm-tools`. Again, `binaryen` will merge two types.

peter-jerry-ye, 2024-04-15:
Seems to be caused by the logic introduced here: https://github.com/wenyongh/wasm-micro-runtime/blob/f5eee3826812351fee8f11f8f8020e04943b51d8/core/iwasm/interpreter/wasm_loader.c#L2101

wenyongh, 2024-04-15:
It should be 

> Hi, I found a counter example here. The following code will have 'function type mismatch' while it shouldn't.
> 
> ```webassembly
> (module
> (rec
>  (type $t1-sig (func))
>  (type $t1 (sub (struct (field (mut (ref null $t1-sig)))))))
> (rec
>  (type $t2-sig (func))
>  (type $t2 (sub (struct (field (mut (ref null $t2-sig)))))))
> 
> (func $f1 (call_ref $t2-sig (struct.get $t2 0 (ref.as_non_null (ref.null $t2)))))
> )
> ```
> 
> It also happens when it's parsed with `wasm-tools`. Again, `binaryen` will merge two types.

Yes, it should be caused by the check for call_ref opcode in the loader, I updated #3317, please help check again. Thanks.

peter-jerry-ye, 2024-04-16:
It fixes the issue. Thanks for the great work.

closed 2024-04-19 by peter-jerry-ye