wasm-micro-runtime/wasm-micro-runtime #3297: mysterious native stack check calculation

yamt, 2024-04-09:
`invoke_native_with_hw_bound_check` and `call_wasm_with_hw_bound_check` have native stack overflow checks
against `native_stack_boundary + page_size * (guard_page_count + 1)`.
i don't understand where this `+ 1` came from.  @wenyongh do you remember?

wenyongh, 2024-04-09:
> `invoke_native_with_hw_bound_check` and `call_wasm_with_hw_bound_check` have native stack overflow checks against `native_stack_boundary + page_size * (guard_page_count + 1)`. i don't understand where this `+ 1` came from. @wenyongh do you remember?

@yamt I remember that it is to reserve some stack space to ensure the runtime can successfully call to the AOT function, for example, avoiding core dump at os_setjmp, since wasm_exec_env_push_jmpbuf has been executed before it, if os_setjmp failed, then in the signal handler, the wrong jmpbuf will be popped for os_longjmp to jump. And we had better let stack overflow occurs in AOT code, not sure whether occurring in runtime part (e.g. wasm_runtime_invoke_native) will cause unexpected behavior.

yamt, 2024-04-09:
> @yamt I remember that it is to reserve some stack space to ensure the runtime can successfully call to the AOT function, for example, avoiding core dump at os_setjmp, since wasm_exec_env_push_jmpbuf [...]

it doesn't seem to make much sense to me because native_stack_boundary at this point already includes WASM_STACK_GUARD_SIZE adjustment, which should be enough for the jmpbuf setup.

> And we had better let stack overflow occurs in AOT code, not sure whether occurring in runtime part (e.g. wasm_runtime_invoke_native) will cause unexpected behavior.

it makes sense.
i'm thinking to separate stack consumption of native functions from WASM_STACK_GUARD_SIZE.

wenyongh, 2024-04-09:
Yes, after some amendings, the code may look a little confusing, please help enhance it if you can.

yamt, 2024-04-25:
fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/3351

closed 2024-04-25 by yamt