wasm-micro-runtime/wasm-micro-runtime #3298: Run wamr with zephyr on nrf5340 board unsuccessfully.

TAOFOR4, 2024-04-09:
### Subject of the issue

```
Building C object CMakeFiles/app.dir/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.obj
FAILED: CMakeFiles/app.dir/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.obj 
ccache /home/tao/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBH_FREE=wasm_runtime_free -DBH_MALLOC=wasm_runtime_malloc -DBH_PLATFORM_ZEPHYR -DBUILD_TARGET=\"THUMB\" -DBUILD_TARGET_THUMB -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF5340_XXAA_APPLICATION -DNRF_SKIP_FICR_NS_COPY_TO_RAM -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DWASM_DISABLE_HW_BOUND_CHECK=0 -DWASM_DISABLE_STACK_HW_BOUND_CHECK=0 -DWASM_DISABLE_WAKEUP_BLOCKING_OP=0 -DWASM_ENABLE_AOT=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWASM_ENABLE_FAST_INTERP=0 -DWASM_ENABLE_GLOBAL_HEAP_POOL=1 -DWASM_ENABLE_INTERP=1 -DWASM_ENABLE_LIBC_BUILTIN=1 -DWASM_ENABLE_MINI_LOADER=0 -DWASM_ENABLE_MULTI_MODULE=0 -DWASM_ENABLE_QUICK_AOT_ENTRY=1 -DWASM_ENABLE_SHARED_MEMORY=0 -DWASM_GLOBAL_HEAP_SIZE=131072 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/tao/wasm-micro-runtime/core/iwasm/interpreter -I/home/tao/wasm-micro-runtime/core/iwasm/aot -I/home/tao/wasm-micro-runtime/core/iwasm/libraries/libc-builtin -I/home/tao/wasm-micro-runtime/build-scripts/../core/iwasm/include -I/home/tao/wasm-micro-runtime/core/shared/platform/zephyr -I/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/../include -I/home/tao/wasm-micro-runtime/core/shared/mem-alloc -I/home/tao/wasm-micro-runtime/core/iwasm/common -I/home/tao/wasm-micro-runtime/core/shared/utils -I/home/tao/zephyrproject/zephyr/include -I/home/tao/zephyrproject/zephyr/wamr/zephyr/simple/build/zephyr/include/generated -I/home/tao/zephyrproject/zephyr/soc/nordic -I/home/tao/zephyrproject/zephyr/soc/nordic/nrf53/. -I/home/tao/zephyrproject/zephyr/soc/nordic/common/. -I/home/tao/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/tao/zephyrproject/zephyr/modules/cmsis/. -I/home/tao/zephyrproject/modules/hal/nordic/nrfx -I/home/tao/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/tao/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/tao/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -isystem /home/tao/zephyrproject/zephyr/lib/libc/common/include -std=gnu99 -ffunction-sections -fdata-sections                                          -Wall -Wno-unused-parameter -Wno-pedantic -mthumb -fno-strict-aliasing -Os -imacros /home/tao/zephyrproject/zephyr/wamr/zephyr/simple/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/tao/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/tao/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/tao/zephyrproject/zephyr/wamr/zephyr/simple=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/tao/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/tao/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT CMakeFiles/app.dir/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.obj -MF CMakeFiles/app.dir/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.obj.d -o CMakeFiles/app.dir/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c.obj -c /home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c: In function 'disable_mpu_rasr_xn':
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:28:16: error: 'MPU_Type' has no member named 'RASR'
   28 |         if (MPU->RASR & MPU_RASR_XN_Msk) {
      |                ^~
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:28:25: error: 'MPU_RASR_XN_Msk' undeclared (first use in this function); did you mean 'MPU_RBAR_XN_Msk'?
   28 |         if (MPU->RASR & MPU_RASR_XN_Msk) {
      |                         ^~~~~~~~~~~~~~~
      |                         MPU_RBAR_XN_Msk
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:28:25: note: each undeclared identifier is reported only once for each function it appears in
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:29:16: error: 'MPU_Type' has no member named 'RASR'
   29 |             MPU->RASR |= ~MPU_RASR_XN_Msk;
      |                ^~
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c: In function 'os_icache_flush':
/home/tao/wasm-micro-runtime/core/shared/platform/zephyr/zephyr_platform.c:227:5: warning: implicit declaration of function 'sys_cache_instr_flush_range' [-Wimplicit-function-declaration]
  227 |     sys_cache_instr_flush_range(start, len);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[120/178] Building C object zephyr/soc/soc/nrf5340/CMakeFiles/soc__nordic.dir/nrf53/soc.c.obj
ninja: build stopped: subcommand failed.

```

### Test case

I use the examples under product-mini/platforms/zephyr

1. Command is 'west build . -b nrf5340dk/nrf5340/cpuapp --pristine -- -DWAMR_BUILD_TARGET=THUMB'
2. I install the zephyr locally instead of using docker. Since running the same code only with zephyr and nrf5340 works well, so I guess my environment is good.

### Your environment

* Linux Ubuntu 22
* My own laptop

wenyongh, 2024-04-10:
@TAOFOR4 Have you updated to the latest code? The similar issue was reported in #3252 and it was fixed with PR #3255 (and #3268 for branch release/1.3.x).

TAOFOR4, 2024-04-11:
> @TAOFOR4 Have you updated to the latest code? The similar issue was reported in #3252 and it was fixed with PR #3255 (and #3268 for branch release/1.3.x).

Hi yes sorry I forgot to update to the latest code. Now it works. Thx

closed 2024-04-11 by TAOFOR4