wasm-micro-runtime/wasm-micro-runtime #3312: Access reference objects in native function

peter-jerry-ye, 2024-04-12:
#### Feature

I'd like to be able to access reference objects in native function with GC proposal (the apis in `gc_export.h` are great). Currently with signature `(r)` only if the type is between `REF_TYPE_NULLREF` and `REF_TYPE_FUNCREF` (btw I'm not sure if the [comparison is correct](https://github.com/bytecodealliance/wasm-micro-runtime/blob/fef26ead3e268768bce3aa658b2a05dbc3b3ee4e/core/iwasm/common/wasm_native.c#L89) ). It would be great if anything between `REF_TYPE_HT_NULLABLE` and `REF_TYPE_NULLFUNCREF` is supported.

#### Benefit

We can integrate better with gc proposal in terms of FFI (for example, no need to serialize/deserialize a structure just to modify a field

#### Implementation

I do not have a plan of implementation yet, and I'm not sure if it possible.

#### Alternatives

N.A.

wenyongh, 2024-04-12:
@peter-jerry-ye yes, thanks for pointing out. Seems it should be `type >= REF_TYPE_HT_NULLABLE && type <= REF_TYPE_FUNCREF`.

peter-jerry-ye, 2024-04-12:
> @peter-jerry-ye yes, thanks for pointing out. Seems it should be `type >= REF_TYPE_HT_NULLABLE && type <= REF_TYPE_FUNCREF`.

Maybe it should be `type >= REF_TYPE_HT_NULLABLE && type <= REF_TYPE_NULLFUNCREF`?

By doing so I get segmentation fault with a value of `REF_TYPE_HT_NON_NULLABLE`. I will try to investigate furthur.

peter-jerry-ye, 2024-04-15:
It solves the issue. Thank you.

wenyongh, 2024-04-17:
Welcome.

closed 2024-04-19 by peter-jerry-ye