wasm-micro-runtime/wasm-micro-runtime #3314: default WASM_STACK_GUARD_SIZE seems too small for linux

yamt, 2024-04-12:
on ubuntu 20.04 amd64, PLT resolution seems to consume about 1500 bytes stack.
thus, the default 1024 is not sufficient for most non-trivial host functions / aot helpers.

wenyongh, 2024-04-17:
Yes, how about increasing it to 2048 for linux platform?

BTW, the size of argv_buf in wasm_runtime_invoke_native seems too large and unnecessary since normally the parameter count of wasm function and host function won't be very large:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/30426be82c806391c9709c5c11c891cdf50b528f/core/iwasm/common/wasm_runtime_common.c#L4883-L4884

How about we also decreasing it, e.g. `uint64 argv_buf[12]`?

yamt, 2024-04-17:
> Yes, how about increasing it to 2048 for linux platform?

sure, it's the simplest solution.

> BTW, the size of argv_buf in wasm_runtime_invoke_native seems too large and unnecessary since normally the parameter count of wasm function and host function won't be very large: [...]

it makes sense to decrease it.
what's your rationale for 12?

yamt, 2024-04-17:
btw, i want to group these issues with a label, say, "native stack overflow detection".
can you give me necessary permissions?

ditto for exception-handling.

wenyongh, 2024-04-17:
> btw, i want to group these issues with a label, say, "native stack overflow detection". can you give me necessary permissions?
> 
> ditto for exception-handling.

OK, done, please have a try.

yamt, 2024-04-17:
> > btw, i want to group these issues with a label, say, "native stack overflow detection". can you give me necessary permissions?
> > ditto for exception-handling.
> 
> OK, done, please have a try.

i can assign existing labels. but i can't create new labels.

wenyongh, 2024-04-17:
> > > btw, i want to group these issues with a label, say, "native stack overflow detection". can you give me necessary permissions?
> > > ditto for exception-handling.
> > 
> > 
> > OK, done, please have a try.
> 
> i can assign existing labels. but i can't create new labels.

I just assigned you with Triage role, creating the new label requires Write role, we need to discuss it internally. Shall I create the "native stack overflow detection" and "exception handling" labels for you first?

yamt, 2024-04-17:
> I just assigned you with Triage role, creating the new label requires Write role, we need to discuss it internally. Shall I create the "native stack overflow detection" and "exception handling" [...]

yes. it's enough.

wenyongh, 2024-04-17:
> sure, it's the simplest solution.
>
>
> it makes sense to decrease it. what's your rationale for 12?

In current libc-wasi, it seems that API wasi_path_open has the maximum argument count: 9 arguments + 1 exec_env = 10, I suppose 12 is big enough for most cases. Or any suggestion from you?

yamt, 2024-04-17:
> > > > > > btw, i want to group these issues with a label, say, "native stack overflow detection". can you give me necessary permissions?
> > > > > > ditto for exception-handling.
> > > > > 
> > > > > 
> > > > > OK, done, please have a try.
> > > > 
> > > > 
> > > > i can assign existing labels. but i can't create new labels.
> > > 
> > > 
> > > I just assigned you with Triage role, creating the new label requires Write role, we need to discuss it internally. Shall I create the "native stack overflow detection" and "exception handling" labels for you first?
> > 
> > 
> > yes. it's enough.
> 
> Done, the two labels were created.

thank you.

yamt, 2024-04-17:
> In current libc-wasi, it seems that API wasi_path_open has the maximum argument count: 9 arguments + 1 exec_env = 10, I suppose 12 is big enough for most cases. Or any suggestion from you?

well, the principle "whatever largest commonly-used api needs" sounds reasonable to me.

however, the actual value seems a bit more larger than that:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/d39d2ba3ca0bdc47ba81e1ee7723e6181fda3b1b/core/iwasm/common/wasm_runtime_common.c#L4910-L4915

yamt, 2024-04-17:
> > Yes, how about increasing it to 2048 for linux platform?
> 
> sure, it's the simplest solution.

probably 2048 is not enough as
aot_call_indirect + invoke_native_with_hw_bound_check + wasm_runtime_invoke_native
seems to consume about 1000 bytes.
how about 3KB?

wenyongh, 2024-04-17:
> well, the principle "whatever largest commonly-used api needs" sounds reasonable to me.
>
> however, the actual value seems a bit more larger than that: [...]

Oh, that's really upsetting, suppose MAX_REG_FLOATS is 8 and SIMD is enabled, allocation will occur when the native API has 8 arguments: 1 + 8 * 2 + 8 * 2= 33 > 32. Maybe 32 is already good.

wenyongh, 2024-04-17:
> > > Yes, how about increasing it to 2048 for linux platform?
> > 
> > 
> > sure, it's the simplest solution.
> 
> probably 2048 is not enough as aot_call_indirect + invoke_native_with_hw_bound_check + wasm_runtime_invoke_native seems to consume about 1000 bytes. (macOS amd64) how about 3KB?

It sounds good to me.

wenyongh, 2024-04-17:
BTW, we are planning to release WAMR-2.0.0 since we haven't released a new version for some time and also there are some developers asking for the new release, is it ok to create the release before your enhancement PR for native stack overflow check is finished? Maybe we can add it in the release 2.0.1.

yamt, 2024-04-17:
> BTW, we are planning to release WAMR-2.0.0 since we haven't released a new version for some time and also there are some developers asking for the new release, is it ok to create the release before [...]

https://github.com/bytecodealliance/wasm-micro-runtime/issues/3325 might or might not involve abi breakage.

closed 2024-04-23 by wenyongh