wasm-micro-runtime/wasm-micro-runtime #3328: How to use coroutines in wasm?

kamylee, 2024-04-17:
I've created a thread to load Wasm, then called the init function within Wasm. This init function invokes native functions to create a TCP server. I'm using the third-party library “coost” to create the TCP server, where methods like accept and recv need to run within coroutine. When a connection is established, a new coroutine is created to execute recv and send operations. However, I'm unable to call functions within the Wasm using exec_env within these coroutines. because they are running in other threads. Even using the wasm_runtime_spawn_exec_env method doesn't resolve the issue. Could you advise on how to call functions within the Wasm in sub-coroutines?

[image]

Are there any other suggestions? thanks!

wenyongh, 2024-04-18:
@kamylee I think your wasm app keeps waiting for message, right? Can you make it run into a loop inside another native api (e.g. Native_WaitMsg)? And NativeWaitMsg maintains a msg buffer or a msg queue to let the server thread fill in the buffer (or send msg to that queue), and in the loop, keep checking the buffer or queue, and then call wasm OnRecv method if a msg comes.

kamylee, 2024-04-18:
thanks! i'll try it now!

closed 2024-04-22 by kamylee