wasm-micro-runtime/wasm-micro-runtime #3329: Add ability to get import and export information from wasm_export interface

bnason-nf, 2024-04-17:
#### Feature

When using the wasm_export API, it currently doesn't seem possible to get information about the imports and exports of a module. When using the wasm_c_api API, that information is exposed. It would be nice to add support for this in wasm_export also.

#### Benefit

Using wasm_export is beneficial in some cases where the added functionality it provides is desired, but if it's not a superset of wasm_c_api then there are drawbacks to using either API.

There is a little more discussion in issue #1740.

#### Implementation

I have a working implementation, and will submit a pull request of that as a starting point for the desired implementation.

#### Alternatives

The only alternative I'm aware of is extremely ugly and unsafe:

```C
    char errorString[128];
    wasm_module_t module = wasm_runtime_load(data, dataSize, errorString, sizeof(errorString));

    char moduleHack[64]; /* sizeof(wasm_module_ex_t) */
    memset(moduleHack, 0xff, sizeof(moduleHack)); /* fill with non-zero to make internal wasm_module_ex_t ref_count not be empty */
    *(wasm_module_t *)moduleHack = module;
    wasm_module_t *module = (wasm_module_t *)moduleHack;

    wasm_importtype_vec_t imports;
    memset(&imports, 0, sizeof(imports));
    wasm_module_imports(module, &imports);
```

bnason-nf, 2024-04-17:
@lum1n0us as we discussed in #1740, I've created this issue and a corresponding pull request. I'm not sure if this matches what you'd like to see stylistically, but it's a pretty simple implementation and fully provides the functionality I'm looking for. I'm happy to make any adjustments you'd like to see.

lum1n0us, 2024-04-18:
> The only alternative I'm aware of is extremely ugly and unsafe:

👍  Can't agree more.

PR #3330 (2024-04-20): Add functions to expose module import/export info
Resolves #3329.
commit 8253c417 (2024-04-20): Add functions to expose module import/export info (#3330)

Resolves #3329.
closed 2024-04-20 by wenyongh