wasm-micro-runtime/wasm-micro-runtime #3336: An unexpected behavior in multiple modes

erxiaozhou, 2024-04-21:
### Subject of the issue
The WAMR compiled  in various configurations do not raise an exception on an invalid case.
### Test case
[memory.init_2164_858_1712663578577073.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15051030/memory.init_2164_858_1712663578577073.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: e7a8b3e743f8821eddbfc526eec19782c51c8408

### Steps to reproduce
1. build WAMR in class interpreter mode / fast interpreter mode / JIT mode / fast JIT mode (Here take JIT mode as an example)
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1   -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```
3. execute the command
```
iwasm --heap-size=0 --llvm-jit -f "to_test" <test_case>
```

### Expected behavior
undeclared function reference (at offset 0x292)

### Actual behavior
No exception

wenyongh, 2024-04-22:
@erxiaozhou thanks for spotting the issue, the check for ref.func opcode in wasm loader is incomplete, I submitted PR #3340 to fix it, please help check and test again.

wenyongh, 2024-04-25:
Close the issue as PR #3340 was merged.

closed 2024-04-25 by wenyongh