wasm-micro-runtime/wasm-micro-runtime #3337: Unexpected Execution ()

erxiaozhou, 2024-04-21:
### Subject of the issue
The WAMR in JIT has unexpected behavior when executing a test case with illegal control flow.
### Test case

[f32x4.gt_55_571_17107906309149652.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15051233/f32x4.gt_55_571_17107906309149652.zip)

[i16x8.extend_low_i8x16_s_7_200_17126751631561782.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15051234/i16x8.extend_low_i8x16_s_7_200_17126751631561782.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: e7a8b3e743f8821eddbfc526eec19782c51c8408

### Steps to reproduce
1. build WAMR in JIT mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1   -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```
3. execute the command
```
iwasm --heap-size=0 --llvm-jit -f "to_test" <test_case>
```

### Expected behavior
 An exception indicating " else found outside of an `if` block"

### Actual behavior
No exception

wenyongh, 2024-04-22:
@erxiaozhou thanks for reporting the issue, I fixed it in PR #3340, please try again.

wenyongh, 2024-04-25:
Close the issue as PR #3340 was merged.