wasm-micro-runtime/wasm-micro-runtime #3345: Error occurs when calling native function with local variable in AOT mode？

kamylee, 2024-04-23:
When I call a native function and pass a local variable, it runs fine in wasm mode. But when I compile it into an AOT file, it crashes when executing wasm_runtime_load. If I use a global variable, it runs fine. Do I need to set something when compiling into an AOT file?

[image]

TianlongLiang, 2024-04-24:
Do you have a minimal reproducible example? Maybe I can give it a try to see where went wrong

TianlongLiang, 2024-04-25:
From you snapshot, I think maybe it's a platform specific question on windows, I tested it on ubuntu 20.04, it seems no error appears with foo2 implemented as following:
```bash
static void
foo2_wrapper(wasm_exec_env_t exec_env, char *name, char *result, int resultlen)
{
    strncpy(result, name, resultlen);
}

static NativeSymbol native_symbols[] = {
    REG_NATIVE_FUNC(foo2, "($*~)")
};
```
I will try it on Windows to see where went wrong

kamylee, 2024-04-25:
I also tried it in Ubuntu 22.04 and it ran okay with local variables.  it seems to be a specific error on the Windows platform. thank you! wait for your conclusion!

Another issue is that when using the printf function to format local variables, it can also crash.

TianlongLiang, 2024-04-26:
Hi, could you please tell me whether you compiled the iwasm and wamrc with the latest commit? I didn't see similar error on Windows

kamylee, 2024-04-28:
the commit date is  2024/3/26 17:55:53
[image]

TianlongLiang, 2024-04-28:
Can you try with the latest commit and recompile both the iwasm and wamrc to see whether the error is still?

kamylee, 2024-04-28:
thank you very much,It's running ok after the update. :)

closed 2024-04-28 by kamylee