wasm-micro-runtime/wasm-micro-runtime #3346: Heap Buffer Overflow Inside "wasm_loader_check_br" Function

mobsceneZ, 2024-04-23:
### Subject of the issue

Running the CLI iwasm with the given testcase results in a heap buffer overflow.

### Test case

[iwasm-poc-01.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15073671/iwasm-poc-01.zip)

### Your environment

```shell
OS               : Linux 5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Commit           : 7bdea3c2ae1f23683299c008bd5093ccaeb5f7b1
Version          : 2.0.0
Clang Verison    : 13.0.0
Affected Tool    : iwasm
Enabled Features : None
```

### Steps to reproduce

```shell
Build            : cd product-mini/platforms/linux/ && mkdir -p build && cd build && export CC="clang" CXX="clang++" CFLAGS="-fsanitize=address -g" CXXFLAGS="-fsanitize=address -g" && cmake .. && make -j
Command          : iwasm -f main iwasm-poc-01
```

### Expected behavior

The program should exit gracefully with possibly some error information.

### Actual behavior

Here is the stack trace provided by AddressSanitizer:
```shell
=================================================================
==4571==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60600000001e at pc 0x55fa640e8694 bp 0x7ffde9e59ee0 sp 0x7ffde9e59690
WRITE of size 4 at 0x60600000001e thread T0
    #0 0x55fa640e8693 in __asan_memcpy /home/lain/llvm-project-llvmorg-13.0.0/compiler-rt/lib/asan/asan_interceptors_memintrinsics.cpp:22
    #1 0x55fa641246c6 in b_memcpy_s /home/lain/wasm-micro-runtime/core/shared/utils/bh_common.c:95:5
    #2 0x55fa641b568d in wasm_loader_check_br /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:10196:13
    #3 0x55fa641b568d in check_branch_block /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:10284:10
    #4 0x55fa6419e2a7 in wasm_loader_prepare_bytecode /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:11582:31
    #5 0x55fa6418b3a3 in load_from_sections /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6001:14
    #6 0x55fa6418eeea in load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6405:13
    #7 0x55fa6418eeea in wasm_loader_load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6582:10
    #8 0x55fa64127bda in wasm_runtime_load /home/lain/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:1386:12
    #9 0x55fa641232d6 in main /home/lain/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:913:25
    #10 0x7fc3a50a2082 in __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:308:16
    #11 0x55fa6405867d in _start (/home/lain/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm+0x4367d)

0x60600000001e is located 2 bytes to the left of 64-byte region [0x606000000020,0x606000000060)
allocated by thread T0 here:
    #0 0x55fa640e975d in __interceptor_malloc /home/lain/llvm-project-llvmorg-13.0.0/compiler-rt/lib/asan/asan_malloc_linux.cpp:129
    #1 0x55fa6418fc58 in loader_malloc /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:378:39
    #2 0x55fa6418fc58 in wasm_loader_ctx_init /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:8061:15
    #3 0x55fa6418fc58 in wasm_loader_prepare_bytecode /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:10785:24
    #4 0x55fa6418b3a3 in load_from_sections /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6001:14
    #5 0x55fa6418eeea in load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6405:13
    #6 0x55fa6418eeea in wasm_loader_load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6582:10
    #7 0x55fa64127bda in wasm_runtime_load /home/lain/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:1386:12
    #8 0x55fa641232d6 in main /home/lain/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:913:25
    #9 0x7fc3a50a2082 in __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:308:16

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/lain/llvm-project-llvmorg-13.0.0/compiler-rt/lib/asan/asan_interceptors_memintrinsics.cpp:22 in __asan_memcpy
Shadow bytes around the buggy address:
  0x0c0c7fff7fb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c0c7fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c0c7fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c0c7fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c0c7fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c0c7fff8000: fa fa fa[fa]00 00 00 00 00 00 00 00 fa fa fa fa
  0x0c0c7fff8010: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0c7fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==4571==ABORTING
```

wenyongh, 2024-04-24:
@mobsceneZ Thanks for reporting the issue! I submitted PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/3352 to fix it, please try again.

mobsceneZ, 2024-04-25:
@wenyongh I checked the PoC again and program exited gracefully now, thanks for your effort!

wenyongh, 2024-04-25:
@mobsceneZ welcome. So let's close the issue as #3352 was merged.