wasm-micro-runtime/wasm-micro-runtime #3347: Out-of-Bound Memory Read inside "block_type_get_arity" Function

mobsceneZ, 2024-04-23:
### Subject of the issue

Running the CLI iwasm with the given testcase results in segmentation fault, which is caused by out-of-bound memory read.

### Test case

[iwasm-poc-02.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15073861/iwasm-poc-02.zip)

### Your environment

```shell
OS               : Linux 5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Commit           : 7bdea3c2ae1f23683299c008bd5093ccaeb5f7b1
Version          : 2.0.0
Clang Verison    : 13.0.0
Affected Tool    : iwasm
Enabled Features : None
```

### Steps to reproduce

```shell
Build            : cd product-mini/platforms/linux/ && mkdir -p build && cd build && export CC="clang" CXX="clang++" CFLAGS="-fsanitize=address -g" CXXFLAGS="-fsanitize=address -g" && cmake .. && make -j
Command          : iwasm -f main iwasm-poc-02
```

### Expected behavior

The program should exit gracefully with possibly some error information.

### Actual behavior

Here is the stack trace provided by AddressSanitizer:
```shell
AddressSanitizer:DEADLYSIGNAL
=================================================================
==13987==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000002 (pc 0x56315c449fbc bp 0x7ffcf1e77630 sp 0x7ffcf1e77100 T0)
==13987==The signal is caused by a READ memory access.
==13987==Hint: address points to the zero page.
    #0 0x56315c449fbc in block_type_get_arity /home/lain/wasm-micro-runtime/core/iwasm/common/../interpreter/wasm.h
    #1 0x56315c449fbc in wasm_loader_prepare_bytecode /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:11552:33
    #2 0x56315c4373a3 in load_from_sections /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6001:14
    #3 0x56315c43aeea in load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6405:13
    #4 0x56315c43aeea in wasm_loader_load /home/lain/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:6582:10
    #5 0x56315c3d3bda in wasm_runtime_load /home/lain/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c:1386:12
    #6 0x56315c3cf2d6 in main /home/lain/wasm-micro-runtime/product-mini/platforms/linux/../posix/main.c:913:25
    #7 0x7f8fd4e1a082 in __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:308:16
    #8 0x56315c30467d in _start (/home/lain/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm+0x4367d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/lain/wasm-micro-runtime/core/iwasm/common/../interpreter/wasm.h in block_type_get_arity
==13987==ABORTING
```

wenyongh, 2024-04-24:
@mobsceneZ Thanks for reporting the issue! I submitted PR https://github.com/bytecodealliance/wasm-micro-runtime/pull/3352 to fix it, please try again.

mobsceneZ, 2024-04-25:
@wenyongh Hi, I manually checked the PoC again and the crash disappeared, thanks for your effort!

wenyongh, 2024-04-25:
@mobsceneZ welcome. So let's close the issue as https://github.com/bytecodealliance/wasm-micro-runtime/pull/3352 was merged.

closed 2024-04-25 by wenyongh