wasm-micro-runtime/wasm-micro-runtime #3349: Error compiling wamr for Windows. Error: invalid numeric argument '/Wextra'

louisvangeldrop, 2024-04-23:
Environment: Windows 11  Visual Studio Build tools => Visual Studio 17 2022

Installing according the installation procedure for Windows:

cd product-mini/platforms/windows/
mkdir build
cd build
cmake ..
cmake --build . --config Release
# ./Release/iwasm.exe is generated

I get the error: 
cl : command line  error D8021: invalid numeric argument '/Wextra' [C:\Users\louis\projecten\wasm-micro-runtime\build\iwasm_shared.vcxproj]

wenyongh, 2024-04-24:
Hi, did you `cd product-mini/platforms/windows` and then `mkdir build` to build iwasm? I found that iwasm_shared.vcxproj is created under `wasm-micro-runtime\build` folder and suspected you create the build folder under wamr's root? In that way, the root's CMakeLists.txt is used instead of `product-mini/platforms/windows/CMakeLists.txt`, and in the former, the `-Wextra` is added in CMAKE_C_FLAGS.

yamt, 2024-04-24:
> Hi, did you `cd product-mini/platforms/windows` and then `mkdir build` to build iwasm? I found that iwasm_shared.vcxproj is created under `wasm-micro-runtime\build` folder and suspected you create [...]

it reminded me https://github.com/bytecodealliance/wasm-micro-runtime/issues/2686

louisvangeldrop, 2024-04-24:
My apologies for this mistake. I didn't cd product-mini\platforms\windows. 
Now it works.

closed 2024-04-24 by louisvangeldrop