wasm-micro-runtime/wasm-micro-runtime #3356: wamrc size_level heuristic

yamt, 2024-04-25:
currently wamrc frontend has some logic to tweak size_level. 
https://github.com/bytecodealliance/wasm-micro-runtime/blob/a36c7d5aa9f8bbbe8da15a0c171fc82fc4ac0729/wamr-compiler/main.c#L566-L589

i guess it's more straightforward to do such a tweak in the backend (eg. aot_create_comp_context) instead because we have a better target knowledge there.
also, considering cross-aot-compiling, it's better to avoid decisions based on macros like _WIN32.

related: https://github.com/bytecodealliance/wasm-micro-runtime/issues/3164

yamt, 2025-04-25:
not really fixed

Septa2112, 2025-05-07:
I'd like to take a look at this issue, but I'm not sure how to reproduce it. Could you share the steps or a bit more context?