wasm-micro-runtime/wasm-micro-runtime #3357: Exception: uninitialized element

DimaSavichev, 2024-04-25:
Hey there. I'm trying to build WASM module, which contains protobuf library, with support of pthread (to be able to terminate it in case of timeout). Without pthread support everything is good and running (also without using protobuf it is okay too). If I add it, module is still built successfully by emscripten, but when I try to use module in c++ code wamr throws "AOT module instantiate failed: Exception: uninitialized element". Can you give any insights into debugging this issue? Is there any way to find out which element is uninitialized and somehow initialize it?

P.S. If it helps I build protobuf by instruction https://github.com/dsyer/protobuf-wasm#building-protobuf

TianlongLiang, 2024-04-26:
Hi, just want to make sure we are on the same page. You can run your application in wasm format, but after it is compiled to aot format, it has the error you describe, right?

DimaSavichev, 2024-04-26:
Not really. I do the following
1. compile my wasm module to .wasm with emscripten
```emcc -std=c++20 -g --no-entry -sERROR_ON_UNDEFINED_SYMBOLS=0  \
    -I$PROTOLIB_PATH $PROTOLIB_PATH/.libs/libprotobuf.a \
    -matomics -mbulk-memory -Wl,--shared-memory,--no-check-features \
    -Wl,--export=__heap_base,--export=__data_end,--export=__wasm_call_ctors \
    -Wl,--export=malloc -Wl,--export=free \
    -Wl,--no-entry \
    example.cpp -o example.wasm```
2. compile .wasm to .aot with wamrc
```wamrc --enable-multi-thread --target=x86_64 --cpu=ivybridge -o example.aot example.wasm```
3. use this module inside my c++ code using wasm_runtime_load, wasm_runtime_instantiate, etc.
Two first steps are finished successfully without errors. But when using module it throws the error described above

TianlongLiang, 2024-04-28:
What if you are loading the wasm file rather than aot file? Will it still have similar errors?

TianlongLiang, 2024-04-29:
If I remember correctly, emcc's pthread support for standalone mode may not be ideal. We have encountered some non-trivial issues about it before. I don't know if you can switch to using wasi-sdk instead, given the protobuf link you posted is compiled with emcc, I think it will be some extra work to use wasi-sdk to compile that library.
PS: When you can get it run successfully, for thrown exception debugging you can refer to [samples/debug-tools](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/debug-tools), which explains how to find the line number of the source code based on the call stack dumped when an exception is thrown.