wasm-micro-runtime/wasm-micro-runtime #3361: Unexpected "type mismatch" Exception on JIT mode

erxiaozhou, 2024-04-25:
### Subject of the issue
The WAMR on JIT mode raises a "type mismatch" exception on a valid test case.
### Test case

[425.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15119467/425.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: f14a166466a2a271493860b6c568bb490b5bd03b

### Steps to reproduce
1. build WAMR in JIT mode
```
export CC=/usr/lib/llvm-16/bin/clang
export CXX=/usr/lib/llvm-16/bin/clang++
cd product-mini/platforms/linux/;rm -rf build
cmake -DWAMR_BUILD_AOT=1 -DWAMR_BUILD_FAST_JIT=1   -DWAMR_BUILD_REF_TYPES=1 -DCMAKE_BUILD_TYPE=Release  -DWASM_ENABLE_BULK_MEMORY=1 -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_SIMD=1  -Bbuild
```
3. execute the command
```
iwasm --heap-size=0 --llvm-jit -f "to_test" <test_case>
```

### Expected behavior
-1
### Actual behavior
An exception 
```
WASM module load failed: type mismatch
```

TianlongLiang, 2024-04-29:
@erxiaozhou Thanks for reporting the issue, this has been fixed in https://github.com/bytecodealliance/wasm-micro-runtime/pull/3374

wenyongh, 2024-05-08:
Close the issue since it was fixed.

closed 2024-05-08 by wenyongh