wasm-micro-runtime/wasm-micro-runtime #3388: Unexpected execution result on fast interpreter mode

erxiaozhou, 2024-05-02:
### Subject of the issue
The WAMR on faster interpreter mode has a wrong execution result on the test case.
### Test case

[tt3.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15186503/tt3.zip)

```
(module
  (type (;0;) (func))
  (type (;1;) (func (param i64 i64) (result i64 i64)))
  (func (;0;) (result i64)
    i64.const 230
    i64.const 366
    i64.div_u
    i64.const 968
    call 1
    i32.const 0
    if (param i64 i64) (result i64 i64)  ;; label = @1
    end
    i64.div_u
    ;; global.get 0
    ;; i64.add
    ;; global.set 0
    )
  (func (;1;) (type 1) (param i64 i64) (result i64 i64)
    i64.const 0
    i64.const 9223372036854775807)
  (global (;0;) (mut i64) (i64.const 853))
  (global (;1;) (mut f32) (f32.const 0x1.2f312cp+3 (;=9.47475;)))
  (export "_start" (func 0))
  (export "to_test" (func 0)))

```

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: 835188cc53373159165423debc48242ad15548d6

### Steps to reproduce
1. build WAMR in fast interpreter mode
2. execute the command
```
iwasm --heap-size=0 -f "to_test" <test_case>
```

### Expected behavior
```
0
```
### Actual behavior
An exception "Exception: integer divide by zero"

wenyongh, 2024-05-08:
@erxiaozhou thanks for reporting the issue, it is similar to #3386 when handling the if block in loader:
```wast
    if (param i64 i64) (result i64 i64)  ;; label = @1
    end
```
Please try #3404.

wenyongh, 2024-05-14:
Close the issue since it was fixed.

closed 2024-05-14 by wenyongh