wasm-micro-runtime/wasm-micro-runtime #3392: There is an issue using sockets in wasm.

kamylee, 2024-05-06:
**code in wasm**：
`
#include "socket_utils.h"
#include <arpa/inet.h>
#include <netinet/in.h>

#include <sys/socket.h>
#include <unistd.h>
#ifdef __wasi__
#include <wasi_socket_ext.h>
#endif

...
int testTcpClient()
{
    int socket_fd, ret, total_size = 0, af;
    char buffer[1024] = { 0 };
    char ip_string[64] = { 0 };
    socklen_t len;
    struct sockaddr_storage server_address = { 0 };
    struct sockaddr_storage local_address = { 0 };

   
    af = AF_INET;
    len = sizeof(struct sockaddr_in);
    init_sockaddr_inet((struct sockaddr_in*)&server_address);

    printf("[Client] Create socket\n");
    socket_fd = socket(af, SOCK_STREAM, 0);
    if (socket_fd == -1) {
        perror("Create socket failed");
        return EXIT_FAILURE;
    }

    printf("[Client] Connect socket\n");
    if (connect(socket_fd, (struct sockaddr*)&server_address, len) == -1) {
        perror("Connect failed");
        close(socket_fd);
        return EXIT_FAILURE;
    }

    len = sizeof(local_address);
    ret = getsockname(socket_fd, (struct sockaddr*)&local_address, &len);
    if (ret == -1) {
        perror("Failed to retrieve socket address");
        close(socket_fd);
        return EXIT_FAILURE;
    }

    if (sockaddr_to_string((struct sockaddr*)&local_address, ip_string,
        sizeof(ip_string) / sizeof(ip_string[0]))
        != 0) {
        printf("[Client] failed to parse local address\n");
        close(socket_fd);
        return EXIT_FAILURE;
    }

    printf("[Client] Local address is: %s\n", ip_string);

    printf("[Client] Client receive\n");
    while (1) {
        ret = recv(socket_fd, buffer + total_size, sizeof(buffer) - total_size,
            0);
        if (ret <= 0)
            break;
        total_size += ret;
    }

    printf("[Client] %d bytes received:\n", total_size);
    if (total_size > 0) {
        printf("Buffer recieved:\n%s\n", buffer);
    }

    close(socket_fd);
    printf("[Client] BYE \n");

    return 1;
}`

**CmakeLists.txt:**
`
...
add_executable (demoS7.wasm "demoS7.cpp" 
	
	)

TARGET_LINK_LIBRARIES(demoS7.wasm 	
	cjson
	socket_wasi_ext
)

set_target_properties(demoS7.wasm PROPERTIES LINK_FLAGS " -O3 \
-Wl,--export=init -Wl,--export=onMessage,--export=onDataRecv \
-Wl,--export=__heap_base -Wl,--export=__data_end \
-Wl,--no-entry -Wl,--strip-all -Wl,--allow-undefined \
")
...
`
**Some error occurred during runtime？**

[image]

[07:02:11:630 - 7FFB3DC9D720]: warning: failed to link import function (wasi_snapshot_preview1, sock_connect)
[07:02:11:010 - 7FFB3DC9D720]: warning: failed to link import function (wasi_snapshot_preview1, sock_open)
[07:02:11:375 - 7FFB3DC9D720]: warning: failed to link import function (wasi_snapshot_preview1, sock_addr_local)

lum1n0us, 2024-05-06:
please refer to https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/socket-api/wasm-src/CMakeLists.txt. 

``` cmake
include(${CMAKE_CURRENT_SOURCE_DIR}/../../../core/iwasm/libraries/lib-socket/lib_socket_wasi.cmake)
```

kamylee, 2024-05-06:
[image]

The problem still persists after I added it.

build iwasm:
cmake .. -DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_PLATFORM=windows
cmake --build . --config Release

lum1n0us, 2024-05-06:
🤔 it's weird. Would u mind uploading full content of your code(including CMakeLists.txt)?

kamylee, 2024-05-07:
> 🤔 it's weird. Would u mind uploading full content of your code(including CMakeLists.txt)?

**Uploaded file:**
[DemoS7.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15228183/DemoS7.zip)

**Command to compile to wasm file:**
`cmake -G "Unix Makefiles" -DWASI_SDK_PREFIX=E:/WorkSpace/DownLoads/wasi-sdk-21.0.m-mingw/wasi-sdk-21.0+m -DCMAKE_TOOLCHAIN_FILE=E:/WorkSpace/DownLoads/wasi-sdk-21.0.m-mingw/wasi-sdk-21.0+m/share/cmake/wasi-sdk.cmake -DCMAKE_SYSROOT=E:/WorkSpace/DownLoads/wasi-sdk-21.0.m-mingw/wasi-sdk-21.0+m/share/wasi-sysroot ..`

there are also some issue with thread:
[image]

kamylee, 2024-05-07:
the same error occurs when usring iwasm.exe:
[image]

lum1n0us, 2024-05-07:
here is a minimum version of CMakeLists.txt:

``` cmake
cmake_minimum_required (VERSION 3.8)

project ("demoS7.wasm")

#已经在官方socket-api项目中生成了libsocket_wasi_ext.a，这里直接链接
include(/workspaces/wasm-micro-runtime/core/iwasm/libraries/lib-socket/lib_socket_wasi.cmake)

add_subdirectory(thirdparty/cJSON)

# 将源代码添加到此项目的可执行文件。
add_executable(demoS7.wasm demoS7.cpp)

target_compile_options(demoS7.wasm PRIVATE -pthread)

target_link_libraries(demoS7.wasm
  cjson
  socket_wasi_ext
  pthread
)

target_link_options(demoS7.wasm PRIVATE
  LINKER:--export=__heap_base
  LINKER:--export=__data_end
  LINKER:--export=malloc
  LINKER:--export=free
  LINKER:--shared-memory,--max-memory=10485760
  LINKER:--no-check-features
  LINKER:--allow-undefined
)
```

And for iwasm building, please use `-DWAMR_BUILD_LIB_PTHREAD=1` instead of `-DWAMR_BUILD_LIB_WASI_THREADS=1` unless using `-DCMAKE_TOOLCHAIN_FILE=wasi-sdk-pthread.cmake` when compiling wasm applications.

BTW, for wasm applications, please use latest wasi-sdk which allows a shorter command like `cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=/opt/wasi-sdk-21.0/share/cmake/wasi-sdk.cmake`

kamylee, 2024-05-07:
wamr has been updated to the lastest version。
wasi-sdk:wasi-sdk-21.0+m
thread run ok after use -DWAMR_BUILD_LIB_PTHREAD=1。
[image]

But the issue with the socket still persists.
win10:
[image]
ubuntu22.04:
[image]

kamylee, 2024-05-08:
The example socket-api has been successfully run：
[image]

However, the example socket API under Windows always fails to compile.
`cmake -DWASI_SDK_DIR=E:/WorkSpace/DownLoads/wasi-sdk-21.0.m-mingw/wasi-sdk-21.0+m -DWABT_DIR=E:/WorkSpace/wamr_source/wabt-1.0.34 ..
cmake --build . --config Release
`
[image]
[image]
[image]

Where is the problem?

lum1n0us, 2024-05-08:
It's a POSIX-like socket. I am not confident about its compatibility with Windows. There might be some requirement.

@eloparco Please help us on socket APIs on Windows.

kamylee, 2024-05-08:
I temporarily resolved this problem using the native socket API, for example, invoking native_connect(), native_send(), and native_recv() instead of the system socket API.

thank you for your help!

closed 2024-05-09 by kamylee